mist-devel/mist-board #5: 1GB HDD image is causing critical system error

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
What steps will reproduce the problem?
1. Attach the HDD image from http://8bitchip.info/atari/DiskImgPP1.html

What is the expected output? What do you see instead?

System should be boot normally to Desktop.
There is a big row of bombs filling the middle of the screen.

What version of the product are you using? On what operating system?

MiST ST core r121.

TOS 1.04, KaosTOS, TOS 2.06, SuperTOS 2.06

Please provide any additional information below.

```

Original issue reported on code.google.com by `ch...@c-64.mobi` on 25 Jul 2013 at 9:11

Attachments:
* [P1110496.JPG](https://storage.googleapis.com/google-code-attachments/mist-board/issue-1/comment-0/P1110496.JPG)

_Copied from original issue: mist-devel/_mist-board_OLD#1_

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

Original comment by `till.harbaum` on 25 Jul 2013 at 11:14

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
Another 80MB (20k 7z file), that boots ok on HATARI.
```

Original comment by `ch...@c-64.mobi` on 25 Jul 2013 at 1:09

Attachments:
* [80.7z](https://storage.googleapis.com/google-code-attachments/mist-board/issue-1/comment-2/80.7z)

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
Hi,

Could it be that the issue is not about the size of the HD image, but rather 
about the driver used ?
I will try with a 1Gb disk image with Hddriver soon.
```

Original comment by `olivierb...@me.com` on 25 Jul 2013 at 6:25

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
Looks like it's a driver issue. The only tested driver so far is ahdi.
```

Original comment by `till.harbaum` on 25 Jul 2013 at 8:44

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
Hi,

I will try to have a look with HDdriver anyway :) And maybe with an image of 
the SD Lotharek provided with the Ultrasatan (ICDPro)

AHDI is a *bit* outdated for my taste ;)
```

Original comment by `olivierb...@me.com` on 25 Jul 2013 at 9:15

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
Can someone post a copy of the HDdriver you have used, guys?
```

Original comment by `krakout` on 26 Jul 2013 at 4:37

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
HDDriver is commercial and cannot be distributed.
Moreover the OP has issue with the PPera image, and it uses its own homemade 
driver.

```

Original comment by `olivierb...@me.com` on 26 Jul 2013 at 6:42

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
Ah, thanks, hadn't realised!
```

Original comment by `krakout` on 26 Jul 2013 at 9:04

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
It doesn't seem to make too much sense to pay 1/4 of the whole MIST board for a 
hard disk driver. 

But i'll see if i can fix the problem. There is a free demo version, perhaps 
that's sufficient to search for the problem.
```

Original comment by `till.harbaum` on 26 Jul 2013 at 2:15

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
Till, the issue of the OP is not with HDDriver !
It's an image with PPera driver. 

Please look at the OP image file, I will check with another image of mine using 
HDdriver.
Note the image I will use is one I use with my real ST
```

Original comment by `olivierb...@me.com` on 26 Jul 2013 at 2:40

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
Sorry for my delayed reply, on my second post its a 80MB hdd image that 
prepared using the HDDRIVER (thax to a friend that provided it to me). This 
image too, causes "bombs" on boot too!
```

Original comment by `ch...@c-64.mobi` on 26 Jul 2013 at 2:53

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
Ok, sorry I had not seen second attachment.
So both have issue ...

I think Till can contact the author of HDdriver for help, he's a nice guy.
```

Original comment by `olivierb...@me.com` on 26 Jul 2013 at 2:56

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
I tried to find the driver used in the OP as a separate package without luck.
```

Original comment by `till.harbaum` on 26 Jul 2013 at 6:18

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
Hi Till, 

For Ppera's driver (original post), you can contact the owner of this page : 
http://atari.8bitchip.info/fromhd.php#DL
The contact is at the bottom (the image of a cat)

I do have a more recent version of the driver Ppera gave me, but it give the 
same error as the OP.
For HDdriver, see with Uwe Seimet here : 
http://hddriver.seimet.de/en/contact.html
```

Original comment by `olivierb...@me.com` on 26 Jul 2013 at 6:21

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
It seems the OP image already crashes when processing the hd images boot 
sector. I extracted and disassembled it. I now need to understand and comment 
it and then try to trace it when running on the board. Perhaps one of you would 
like to help and comment the attaches source :-)

Has anyone ever tried to copy the 1GB.img contents (which are slightly over 
128MB only) onto a small ahdi image?
```

Original comment by `till.harbaum` on 26 Jul 2013 at 6:54

Attachments:
* [1GB_boot.img.s](https://storage.googleapis.com/google-code-attachments/mist-board/issue-1/comment-15/1GB_boot.img.s)

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
I did it before, but on real machine. Most games adaptations works.
Will try it again but with MiST. However, as AHDI works, it should work :)
```

Original comment by `olivierb...@me.com` on 26 Jul 2013 at 7:10

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
!! SUCCESS (partial) !!

I have to say that I just got a 200MB, configured with HDDRIVER.
The difference is this image was prepared on STEEM not under HATARI!
I noticed that HATARI has some issues whie booting floppy disks, could not load
some prg's under AUTO folder then I realized that something is wrong/broken 
there!
The issue with PPera's still exist, but we can use at least a bigger HDD made 
with the HDDRIVER, but the price of the drive is too much...
```

Original comment by `ch...@c-64.mobi` on 26 Jul 2013 at 7:32

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
Hi,

Nice !
What is your Hddriver version ?
What is your Hatari version ? Maybe it's related to the version you are using ?

(sorry If I ask too much questions, I work in QA..... ;) )
```

Original comment by `olivierb...@me.com` on 26 Jul 2013 at 7:35

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
I have to ask my friend, I don't have the HDDRIVER, but as I can see on the 
boot message (too fast) is 8 something... the hatari is 1.7.0 as I told him to 
get the  latest version. I personally don't not use the Steem because of my 
system (win8 x64) there is no fullscreen
Anyway, I read there is another driver called CBHD that is freeware.
I will give a try too under steem. 

```

Original comment by `ch...@c-64.mobi` on 26 Jul 2013 at 8:39

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
[deleted comment]
```

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
Now a strange idea came into my mind, could PPera use the STEEM to make the 
image..

(edit/ typo)
```

Original comment by `ch...@c-64.mobi` on 26 Jul 2013 at 9:01

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
Another successful test!

I got now the ICD pro driver (v.6.55) (posted on AtariAge forum).
Well, it seems the problem is *only* with PPera's image.
A 500MB HDD image made with ICDpro (under Steem, as the HATARI refuses to boot 
it)
works without any problem under MiST!

I think you can close the issue, as its definitely a bug with PPera's driver.
I did not check the CBHD, but I am sure it will work.

Thanks.
```

Original comment by `ch...@c-64.mobi` on 26 Jul 2013 at 9:22

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
Great! Thanks for the feedback.

I'll look into this once more as it definitely is interesting. I see the ppera 
image loading the boot sector of the image and then throwing bombs. I'd at 
least like understand why that happens.
```

Original comment by `till.harbaum` on 27 Jul 2013 at 7:20

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
I'm Ppera, and must say that there is not much logic in what is told in many 
posts here.  Bug in driver ?  Which shows only with new HW clone, Mist ?  Why 
bug happens not on ST, STE, TT ?  Fact that some (older) drivers like AHDI work 
means not that others are buggy. There could be diverse reasons - like DMA 
timings or issues in ACSI-SD card transfers.
Throwing bombs at autoboot seems like unreliable reading or reading of bad 
sector, or not reading it at all - from some reason.
My driver is made according to specs from Atari, according to Atari ST 
ProfiBuch DMA description.
So, I would suggest to improve emulation.
Considering Hddriver:  there is free demo which can serve well for testings.

And it would be good to test with non-autoboot Ppera driver - contact me for it 
and more at:   petari@8bitchip.info  .
```

Original comment by `pput...@freemail.hu` on 4 Aug 2013 at 7:50

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
@Ppera:

Did you make this image on Hatari? I am just asking, as I noticed that I could 
not boot on MiST any HDD image that I made on Hatari. Could you please try to 
make this, or a smaller , image (just for testing) on Steem (preferable v 3.2 
not the SSE).
thx
```

Original comment by `ch...@c-64.mobi` on 4 Aug 2013 at 12:45

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
I made image with Steem 3.2 .  But I don't think that it is so relevant. I can 
do it on real SD card and Atari too, but usually do with Steem because 
practical reason:  always start with and empty image file, so it will be better 
RAR-able than some SD card image, what was already filled up to end, so there 
will be need to pack lot of deleted files (which content is still on drive, 
even after repartititoning).

I would like to see some detailed specs about Mist ACSI-SD part.
Only found this:  "  SD cards

Standard form factor classic SD cards (<= 2GB) as well as SDHC cards (> 2GB) 
are supported. The use of a 1GB card of a well-known brand is recommended.

The card has to be FAT formatted. If in doubt use a windows PC or even better a 
device like a digital camera to format the card. "

It seems that SD card is used mainly as floppy image holder ( therefore must be 
FAT formatted ? ) .  Maybe hard disk image must be written not straight at 
start of card ?  I really have no clue how Mist knows is content of SD card 
hard disk image , or floppy images in special format ...  Someone knows more 
about ?

```

Original comment by `pput...@freemail.hu` on 5 Aug 2013 at 6:57

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
Mark the issue as SOLVED.  Ppera's image works JUST perfect using EMUTOS, seems 
a bug definitely with original TOS.
```

Original comment by `ch...@c-64.mobi` on 12 Nov 2013 at 8:27

harbaum, 2015-11-26:
Fixed a loooooong time ago ....

closed 2015-11-26 by harbaum