mist-devel/mist-board #6: [Feature req.] Keyboard Shortcuts

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
If its possible, please:
Keyboard shortcuts for

Reset 
Cold Reset
Change Monitor Mode (color /mono)
NMI (usefull for debug/code monitors/rippers)

Can the AltGR + any other key be mapped for this task?
```

Original issue reported on code.google.com by `ch...@c-64.mobi` on 25 Jul 2013 at 2:50

_Copied from original issue: mist-devel/_mist-board_OLD#2_

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
Some TOS versions have these shortcuts built-in like Ctrl-Alt-Del and 
Ctrl-Alt-RShift-Del for reset and cold boot.

Is the AltGr really present on all keyboards? Even on american ones?
```

Original comment by `till.harbaum` on 25 Jul 2013 at 8:27

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
yeah, you are right.. seems the c-alt-del works with a lot of games I tried
so please ignore the reset shortcuts recuest as its handled by the TOS.

the US keyboard use the AltGR (right Alt) for symbols like ¼ ½ ¾ etc.

my idea about the NMI key is for use with debuggers etc, I had in mind the 
multiface cart (sadly this cart is not working with the MiST)
(cart image can be send by request on email , as I am not sure about the 
copyrights)
but as this cart does not working here, no problem.
```

Original comment by `ch...@c-64.mobi` on 25 Jul 2013 at 8:50

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
Also, I think the disk selection sub-menu (on the OSD) needs a Pageup/Pagedown 
if possible, to speed up browsing the disk images.

```

Original comment by `ch...@c-64.mobi` on 26 Jul 2013 at 4:04

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
Couldn't agree more about the page up/down in the OSD list... Right now it's 
very slow to navigate!
```

Original comment by `krakout` on 1 Aug 2013 at 1:43

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
Are you aware that floppy disk images can be stored in subdirectories? This 
might help avoid huge directories. Nevertheless, pageup/pagedown in the file 
selector is indeed useful.
```

Original comment by `till.harbaum` on 4 Aug 2013 at 7:42
* Changed state: **Accepted**

rkrajnc, 2015-08-05:
_From @GoogleCodeExporter on July 30, 2015 20:44_

```
I do have them in subdirectories. But it's not that practical, of course.

Take Automation, for instance. It's 576 menus, many with multiple disks and 
multiple editions. 2000 files in all. If you put 50 files in a dir you need 40 
of them just to cover Automation... and it's still too slow (do try and see how 
slow it is with long filenames... you need a second just to go two files down!).

FInd-as-you-type is a real winner, too. The most efficient method, probably...
```

Original comment by `krakout` on 4 Aug 2013 at 7:55

harbaum, 2015-11-26:
PageUp and PageDown works. Also you can press a character and jump to the first entry beginning with that character.

closed 2015-11-26 by harbaum