mist-devel/mist-board #7: "Panic: Bus Error." with new MIST and Atari ST emulation

hanshuebner, 2015-11-04:
Hi,

I've just received a new MIST from Lotharek and am trying to follow the instructions on https://github.com/mist-devel/mist-board/wiki/GettingStarted to get the Atari ST emulation running.  When I start the box, I see this message:

```
Panic: Bus error.
misc = 2314, address = 00e02314, opcode = 2314, sr = 2314, pc = 00e06e82
Dregs: 0000ff00 000003e8 00000000 00000000  0000ffff 00000000 00000000 00000000
Aregs: 00000000 00000522 00004a4c 00000000  01000000 00000200 00e00b38 00000ede
 usp = 00000000
````

It seems that the m68k is running, but somehow crashes.  I've tried both the files linked from GitHub as well as the files present in the Subversion repository.  I've tried a few of the other bitstreams (colecovision, spectrum, Amstrad), and they all seem to work to some degree.  Does anyone know whether this is a hardware or software issue?

Thanks!
Hans

renaudhelias, 2015-11-05:
Hi Till,

By the way,

Here a HOWTO about adding/merging CRC16 into MiST-board core firmware :)

For Amstrad dsk access robustess I make my own firmware (but it is not a official one, just running for CoreAmstrad, and failling perhaps in "firmware update" OSD task), Till you can try to merge it, inserting it firstly with a general "boolean switch" to activate/desactivate this functionnality, adding slowly and friendly this purpose to your main firmware core.

CRC16 is activated adding some post-instruction in mmc.c MMC_Init()
http://github.com/mist-devel/mist-binaries/blob/master/cores/amstrad/amstrad_150729_r003.10.crc16.firmware.zip

In mmc.c, do look after "Calculate CRC16 CCITT", it defines two main functions about computing a CRC16 for sdcard SPI protocol :
http://github.com/mist-devel/mist-binaries/blob/master/cores/amstrad/amstrad_150729_r003.10.crc16.firmware.zip

In the other hand, in sd_card.v, do look after "CRC16_WRITE" you should turn into 1 value (instead of 0) in others project than CoreAmstrad to activate compute of CRC16 in verilog way (in Amstrad the CRC16 is computed further : in my own bootloader component)
http://github.com/renaudhelias/CoreAmstrad/blob/master/BuildYourOwnZ80Computer/sd_card.v

I know that my firmware fork does run fine with Amstrad sdcard read/write, but not with MIST core (as sd_card.v is not merged with my own one)
I didn't succeed in compiling and running last official firmware, so I let you make the stuff Till if you want (first step merging with a switch boolean, second step trying it)

Regards,

Renaud (CoreAmstrad)

hanshuebner, 2015-11-05:
Following Till's advice, I tried using TOS 2.06 from http://www.atariworld.org/tos-rom/ and with that, the ST core booted fine.

closed 2015-11-05 by hanshuebner

harbaum, 2015-11-08:
I still need to integrate Renauds changes. But before i need to make sure these changes don't break any of the other cores which also use the SD card.