mist-devel/mist-board #9: bbc_mist_top.qpf

fordp2002, 2015-11-25:
Hi I just tried to open the BBC project file and it says it has errors on my Quartus II 13.0.1.

Any ideas how to proceed. I want to prove I have I can build an exact copy of the project sof file before I start to make improvements.

Cheers.

FordP

harbaum, 2015-11-26:
There was the supermmc rom/hex file missing and the project file missed a reference to the adc.v. I just fixed both. Please retry.

Btw: The latest quartus working for the MIST board is 13.1.4. But 13.0.1 should also work.

fordp2002, 2015-11-26:
Thanks for the fix I will report back in due course.

Awesome project!

I plan on adding Midi support to the BBC Micro by the way.

I will check that Quartus version as I have a DE1 too so need to be sure it is compatible with that!

Cheers.

FordP

fordp2002, 2015-11-27:
I am still having issues:
Error (125048): Error reading Quartus II Settings File C:/Users/Simon/Documents/GitHub/mist-board/cores/bbc/fpga/mist/bbc_mist_top.qsf, line 137
	Info (125063): set_global_assignment -name ENABLE_CONFIGURATION_PINS OFF
Error (125048): Error reading Quartus II Settings File C:/Users/Simon/Documents/GitHub/mist-board/cores/bbc/fpga/mist/bbc_mist_top.qsf, line 138
	Info (125063): set_global_assignment -name ENABLE_NCE_PIN OFF
Error (125048): Error reading Quartus II Settings File C:/Users/Simon/Documents/GitHub/mist-board/cores/bbc/fpga/mist/bbc_mist_top.qsf, line 139
	Info (125063): set_global_assignment -name ENABLE_BOOT_SEL_PIN OFF
Cheers.

FordP

sleary78, 2015-11-27:
You need QuartusSetupWeb-13.1.0.162 NOT 13.0.1

S.

sleary78, 2015-11-27:
You also need the cyclone III packages installed.

S.

fordp2002, 2015-11-27:
Thanks for your help.
I will need both versions I guess as I have a DE1 and 13.0.1 seems to be the last version that supports Cyclone II.

I am downloading now.

Ho Hum, thanks for you patience!

FordP

sleary78, 2015-11-27:
I have mine in a Linux VM. I think it will all build off the make files.

fordp2002, 2015-11-27:
Thank you for your help I have it building now so this issue is officially closed!

Cheers.

FordP

closed 2015-11-30 by harbaum