mist-devel/mist-board #11: BBC Micro SDHC Support

fordp2002, 2015-11-30:
The BBC core claims to not support SDHC but I think this is wrong have read the documentation.

The mmb file always needs to be the first file added to the SD card so should always lie in the first 2GB and therefore meet the requirement for SDHC Support.

I will try and get round to more testing soon and report back. (As well as having a go at adding Midi to the core).

Thanks for this awesome project!

FordP

harbaum, 2015-11-30:
SDHC requires a few more things over normal SD. It needs to be addressed block wise instead of byte wise etc. The IO controller on the MIST board can do some magic and try to translate between these. But that's dangerous as this will only work in the first 2GB of a SHDC card and will lead to problems with an SD card containing more than 2GB.

However, if the supermmc rom is being replaced by something that supports SDHC natively then the core can be told not to do this translation and as a result SDHC cards will work nicely. But imho there is no supermmc like rom supporting SDHC.

IMHO Hoglet is doing something like that here: https://github.com/hoglet67/BeebFpga/tree/master/src/mmfs

The original beemmc which this core also implements uses some bitbanging approach to access the sd card through the user port. If we have the source code of the driver (e.g. the supermmc replacement) then we could easily implement some simple hardware SPI for faster SD card access.

renaudhelias, 2015-11-30:
Last time I had such of "SD not supported" message in my project, it was about a bad plug (a bad wire mapping) of I/O between data_io.v component and sdcard_io component and general schematic in/out...
About 2Go, you normaly just have to defrag (if < 2Go). Around 2Go problems, becare about using some "integer variable" that are using 32bit for address (so 2Go ?)

fordp2002, 2015-11-30:
I will do some testing and get back to you as I have a DE1 and am testing Hoglet's core too.

The mmb file always has to be the first one on the card however so I think with your translation software either the old or the new MMC rom would work fine.

None of the implementations use pure bit bash by the way, they use the shift registers in the 6522 to hardware assist the SD/ SDHC access.

Cheers.
FordP

sorgelig, 2016-01-14:
@harbaum,
Correct me if i'm wrong, but SDHC flag is always checked in current FW regardless if it's real or virtual SD card. 
Thus, i think the requirements can be greatly relaxed for cores when using virtual SD card. Core may have no support for SDHC card, but for virtual SD card it doesn't matter as long as size of virtual image is bellow 4GB (even being on SDHC card) which for retro machine is always true.

harbaum, 2016-01-14:
That's indeed true. The only problem is accessing a real sdhc card like a normal sd card.

gyurco, 2018-09-13:
I think this can be closed, as the core switched to MMFS, so it supports SDHC (considering MMFS limits).

closed 2018-09-13 by gyurco