mist-devel/mist-board #12: C64 source code is missing.

sorgelig, 2015-11-30:
Hi,

Patches in C64 core require vhdl_c64_c1541_sd.zip which is replaced by newer version and thus cannot be applied.
Can anyone upload vhdl_c64_c1541_sd.zip or already patched sfor MIST source code?

harbaum, 2015-11-30:
Yupp, seems the source is gone. We only have permission to redistribute the binaries but not the code. So i cannot help here.

closed 2015-11-30 by harbaum

sorgelig, 2015-11-30:
Is it possible to re-base the patch to new source code? I believe you already have old source code. 
This should be under permission as well.

sorgelig, 2015-12-01:
Actually, new source is compatible. There is one patch chunk failed but it meaningless. So, if anyone want to get working C64 source code for MIST, then hurry up :)