mist-devel/mist-board #13: ZX Spectrum: OSD is not correct in TV RGBS mode.

sorgelig, 2015-12-01:
With scandoubler_disable=1 in mist.ini, OSD position is running vertically.

sorgelig, 2015-12-01:
Fixed by myself :)

```diff
From 06f747fc3ec67a1febe558630847047dc6e27a40 Mon Sep 17 00:00:00 2001
From: sorgelig <pour.garbage@gmail.com>
Date: Tue, 1 Dec 2015 21:55:58 +0800
Subject: [PATCH] Fix OSD menu in RGBS mode.

---
 spectrum_mist.vhd | 2 +-
 video.vhd         | 2 +-
 2 files changed, 2 insertions(+), 2 deletions(-)

diff --git a/spectrum_mist.vhd b/spectrum_mist.vhd
index a5c714d..fe4006d 100644
--- a/spectrum_mist.vhd
+++ b/spectrum_mist.vhd
@@ -1295,7 +1295,7 @@ begin
 	zx_green <= vid_g_out & "00";
 	zx_blue <= vid_b_out & "00";
 	VGA_HS <= vid_hcsync_n;
-	VGA_VS <= vid_vsync_n;
+	VGA_VS <= '1' when scandoubler_disable = '1' else vid_vsync_n;
 
 	-- route video through osd
 	osd_d : osd
diff --git a/video.vhd b/video.vhd
index 1c2189a..5fc0df5 100644
--- a/video.vhd
+++ b/video.vhd
@@ -124,7 +124,7 @@ begin
 	
 	-- Output syncs
 	-- drive VSYNC to 1 in PAL mode for Minimig VGA cable
-	nVSYNC <= '1' when VGA = '0' else not vsync;
+	nVSYNC <= not vsync;
 	nHSYNC <= not hsync;
 	nCSYNC <= not (vsync xor hsync);
 	-- Combined HSYNC/CSYNC.  Feeds HSYNC to VGA HSYNC in VGA mode,
-- 
1.8.3.msysgit.0 
```

harbaum, 2015-12-01:
Excellent. Thanks a lot. Such contributions really make life easier. I just send you an invitation. Please feel free to check in your changes yourself.

harbaum, 2015-12-03:
fixed

closed 2015-12-03 by harbaum