mist-devel/mist-board #14: install_arm-none-eabi-gcc.sh is not complete

sorgelig, 2016-01-02:
Building of gcc failed because required libs missed: MPC, MPFR, GMP
I believe some linux distro have these libraries installed and thus gcc will be built successfully, but for general case it's better to download these libs and put into gcc source folder - and they will be automatically picked by compiler. 
That's how i've compiled it in Ubuntu 14.1.  

I hope i won't need ARM JTAG soon ;)

harbaum, 2016-01-02:
That should just be a matter of apt-get installing the apropriate -dev packages. That works at least for all the ubuntu versions I've used so far incl 15.10.

The arm flasher nowadays also needs a small fix. I should actually document these.

harbaum, 2016-01-02:
Added some remarks regarding the libraries to 
https://github.com/mist-devel/mist-board/wiki/HowToCompileTheFirmware

closed 2016-01-02 by harbaum