mist-devel/mist-board #15: 15Khz support for NES core

amigrave, 2016-01-03:
15Khz support for the NES core would be wonderful because I could get a beautiful RGB output instead of the dull RF of the real hardware.

I can see that it is done using the `scandoubler_disable` parameter but I currently have absolutely no experience with fpga ( *cough* neither electronics *cough* ) so I don't have any idea about where to start.

Could you give some hints ?

harbaum, 2016-01-03:
That's not trivial as the NES core was designed to only support VGA and has the scan doubler built very deep into its design.

If it was easy enough to just explain the necessary changes then i'd actually just quickly do it myself. But it's not that easy.

harbaum, 2016-01-03:
There are a few tutorials at https://github.com/mist-devel/mist-board/tree/master/tutorials if you just want to get started looking into FPGA programming. But these won't teach you VHDL or Verilog in general. You'd need a seperate tutotial for that.

amigrave, 2016-01-05:
Thanks for the link. Indeed I have a lot to learn before being able to tweak those modules.

BTW, do you think that I could "undo" the scan doubler [using one of those converter](http://www.amazon.com/Andoer-Portable-GBS-8100-Definition-Converter/dp/B00XKWX4F8) or will it mess up the output ?

harbaum, 2016-01-05:
I have a similar even cheaper device. It works but the image quality is not very impressive after all those conversions.

Someone should really try to add 15khz support to the nes core.

sorgelig, 2016-01-05:
I'm interesting in NES core as well, and my config requires 15khz (PAL/NTSC RGBS) mode. I will try to make original video mode for this core when i will finish my BK0011M core :)

renaudhelias, 2016-01-05:
I saw a NES modeline by here : www.geocities.ws/podernixie/htpc/modes-en.html
It's clock multiplied by three gives 16MHz.

sorgelig, 2016-01-06:
well.. it wasn't hard to make TV RGBS output.
I will figure out how to make switch using scandoubler_disable option and will release update.

Newsdee, 2016-01-06:
Maybe I'm splitting hairs but I think other cores have 15khz on RBGHV.
Maybe we need an extra option to choose the type of sync?

sorgelig, 2016-01-06:
Nope. TV modes (PAL and NTSC) use RGBS in all cores because TV sets require composite sync.
RGBHV is only for VGA modes to use with monitors.

Newsdee, 2016-01-06:
VGA 240p is a valid mode :)

My VGA->SCART setup seems to run fine with all cores on 15khz except for
the Atari 800/5200. I thought it was because that core implements proper
RGBs, and assumed the other cores were doing VGA (RGBhv) at 15khz.

My understanding vould be wrong of course. Perhaps there is something
peculiar in the 800 core that trips the UMSA chip, whereas the others pass
through the sync signal fine.

Not a big deal, just an observation.

sorgelig, 2016-01-06:
I've pushed commits with 15kHz mode added.
VGA mode is also available. Switch the modes made as usual by scandoubler_disable in mist.ini.

here is precompiled binary for tests:
http://www.mediafire.com/download/sv013pu3ykhd3q5/nes.rbf

If it works fine, then Till can update the binary in repository.

sorgelig, 2016-01-06:
In ideal solution is to have standalone scaler chip right on MIST board. It can be another FPGA with its own independent RAM. And 2 connectors: VGA (for all types of analog output) and HDMI. Or simply single DVI-I :)
In this case we could control what resolution and refresh rate to out and have TV, VGA, YPbPr, HDMI on demand.
That would be an ideal solution!  Till, We need MIST v2 board ;)

sorgelig, 2016-01-06:
Or just separate scaler board based on FPGA, which VGA input and HDVI+VGA output.
I've tried several scaler boards and all of them simply rubbish. They usually accept either only TV or VGA. There is no universal scaler board. I hoped DBS8200 board will solve the problem, but it turned out the same rubbish.
But having FPGA scaler board we could write good firmware to cover all possible video modes.
I would like to buy such board.

harbaum, 2016-01-06:
Wow ... now THAT was fast. Cool! Thanks a lot, will test asap.

An external scan doubler would be as expensive as the mist itself. I think the current solution is fine for most cases. A single line scan doubler we usally use is usually sufficient and only needs very few resources. Spending an entire fpga for that would be overkill. A bigger mist could hace a bigger fpga and a second ram chip. But I actually would like mist v2 to be smaller and cheaper than the current one.

renaudhelias, 2016-01-06:
Using a VRAM chip instead of SDRAM chip, we should reach <20ns and then simulate more RAMs/ROMs ?

MmmPT, 2016-01-06:
As a consumer, let me say that i am tired of fpga Systems without hdmi.

I don't have a vga monitor and mist only works for me with the Amiga core. It would be more cheaper to buy an Amiga 1200.

I would be more than glad to buy a system with hdmi connection.

Regards

Miguel

sorgelig, 2016-01-06:
Single line scan doubler cannot solve one important thing: frame rate conversion. 50Hz is not a VGA mode. So you are very limited where you can use it. 
My home theater doesn't support this scan doubler. Also, i live in area where everything is NTSC or HDTV 60Hz. So, no chance to use even PAL without scaler. 
Thanks, i've found one SCART->HDMI scaler supporting frame rate conversion and output it in 1280x720@60p. But it doesn't support VGA modes. So i'm limited to PAL/NTSC output only. And this scaler has strange problem - my Yamaha Receiver only show picture from this scaler if i open receiver's menu. 
Well, there are so many problems with those scalers, as you see.. So, i'm ready to buy FPGA Scaler for the same price as MIST board just to close this topic once and forever :)

How you can made v2 board cheaper? Only to cut something, i guess... Cheaper FPGA? And there will be chance that some cores won't fit. 

I would prefer SRAM ;) And there will me more time for shared access for RAM. May be even can add normal scaler with frame rate and resolution conversion ;) But SRAM should be very expensive...

sorgelig, 2016-01-06:
I definitely agree, the MIST board should have HDMI. It would sole a lot lot lot problems. I don't have any device with VGA input. So i have to add scalers, converters...

sorgelig, 2016-01-06:
Back to NES core.
I think, this core need more work for keyboard support. Currently core works only in joystick mode which is not convenient, IMHO. I think, keyboard can simulate joystick directly without need to switch the mode. Keys like 1, 2 can switch between joystick 1 and 2.

harbaum, 2016-01-06:
Hdmi causes more problems than it solves. With hdmi all cores would have to use a real vga mode since hdmi typivally doesn't accept those 50hz video modes. So you always need full frame conversion incl. a frame rate conversion. But then video would not be in sync with the core anymore and synchonous video effects don't work anymore.

Many games e.g. scroll at exactly 50hz. Converting this to 60hz results in tearing and stuttering. But then you could use an pc with an emulator as these have the same video problems. Frame rate conversion is not trivial. You'd need to interpolate between frames and even then it looks bad. Even holywood often fails when converting between cinema and various dvd and blueray formats.

And hdmi costs license fees.

sorgelig, 2016-01-06:
HDMI has standard PAL mode 576@50i/p. For those having PAL TVs, no need any conversion.
As for tearing: as i've told before, i'm using PAL->60P scaler, and didn't notice any artifacts. Most retro computers didn't tied to vsync and perfect smoothness. You are exaggerating these conversion problems :) May be only Amiga can show the difference in conversion. But Amiga has NTSC version.

For license fees: may be there are some workarounds. I've saw some similar FPGA boards (Altera MAX10) with HDMI with price even lower than MIST. I believe, if you will use DVI-I, it won't require any fee. And DVI-I is better because can have both VGA and HDMI signals. Users can buy (or get from old video card) cheap (<1USD) DVI-HDMI or DVI-VGA adapter. That will be ultimate universal solution. Don't you agree?

Newsdee, 2016-01-06:
The XRGB-Mini is a great FPGA upscaler, but then it costs much more than
the MiST :)  I use it to convert 50hz to 60hz among other things.

Cheaper converters seem to handle 60fps signals better than 50hz. So
getting a "fully compliant" 60hz VGA signal from all cores could be a first
step without needing MiST hardware changes.

harbaum, 2016-01-06:
I wasn't aware that hdmi supports 576i. That would indeed be fun. But i assume only TV sets can do that? My tests with PC screens with hdmi/dvi were quite bad and compared the VGA input on the same screen i was surprised how much better the VGA input coped with the non-standard video signals.

And yes, it's the Amiga and the Atari ST that have the super-smooth 50Hz scrolling and similar. But that's what 90% of the users use.

Anyway, hdmi/dvi is definitely an option for the next MIST board as analog VGA is already a little retro by itself.

Newsdee, 2016-01-06:
Would be great if more mappers could be added to the NES core. MMC4 for
example would allow using Famicom Wars and two Fire Emblems. Only three
games, but good ones :)

sleary78, 2016-01-06:
Happy to assist with a MiST V2.0. Xilinx chips had native support for hdmi/dvi (cough) ;)

sorgelig, 2016-01-06:
Yeah, 576i is for TV sets. I think, monitors support only PC subsets of HDMI resolutions.

Just pushed commit for keyboard support in NES.
1 - switch to joystick A
2 - switch to joystick B
Up,Down,Left,Right
Esc - start
Tab - select
Space - fire 1
Left Alt - fire 2
If there are better buttons map then let me know :)

Unfortunately i don't know about mappers and how hard they are to implement. Any good links about it are welcome.

Newsdee, 2016-01-06:
Happy to help on testing and digging out docs for mappers. I've done a
start on the forums: http://www.atari-forum.com/viewtopic.php?f=101&t=29120

harbaum, 2016-01-06:
Most consoles and home computers do 288p instead of 576i.

harbaum, 2016-01-06:
The boot menu is usable on my TV in 15kHz mode. But whenever i load a game the video output stops. I'll try another screen this evening.

But you shouldn't hesitate to upload such versions to the official repository. The only thing we have to care a little more about are binary releases of the firmware as it's a little more tricky to reflash if the installed firmware is broken and unable to update itself.

But cores and and source code (core or firmware) are uncritical and i prefer to have a broken core (which wouldn't hurt since the old versions are still available) than missing some cool new feature. So please once you think a new feature is ready for testing, then please release a binary core the official way.

sorgelig, 2016-01-06:
May be HSync/VSync need more tweaks. Need to check their times in 15khz mode. 
I'm testing with my SCART->60P scaler which works fine with new NES NTSC TV mode.
I've tried Mega Man III, Mario Bros 3, Boulder Dash. All of them are NTSC. 
I didn't try PAL games. Do they even supported by this core?

sorgelig, 2016-01-06:
288p is a virtual resolution. When retro computers/consoles output to TV, they use 576i where each field has 288 lines. Thus 576i is native TV resolution for all retro PAL consoles/home computers.

sorgelig, 2016-01-06:
I've just found several scan lines were missed in beginning of frame in 15kHz mode. I've pushed commit, but didn't release binary.
I think, video module requires more cleanup before release.

amigrave, 2016-01-06:
@sorgelig Thanks a lot for your work ! I'd like to test your previous binary, but I wonder if a bug could burn the Sony PVM I just bought. Do you think it's a plausible scenario or is it 100% safe to test your binary ?

harbaum, 2016-01-06:
@amigrave Of course noone will guarantee your anything. You PVM might just coincidentally burst into flames when driven by the nes core.

But i'd say that a high end professional crt should for sure copre with a broken tv signal. The things were built to be driven by test signals and for that reason should not be affected by wrong frequencies.

sorgelig, 2016-01-07:
Was fighting with strange bugs in OSD code. Pulled almost all my hair out :)
It turned out Quartus didn't like horizontal sync as a clock! Now, after fix for this problem, NES core is ready for 15kHz mode.
Binary has been pushed to binaries repository.

harbaum, 2016-01-07:
I don't get a stable image neither on my TV nor on my test screen (which is a NEC VGA LCD which can display 15khz). 

And yes, sync_h cannot be used as a clock as it's "dirty". It's generate by a combinatorial equation from a counter. For a "normal" programmer this looks fine. But in hardware there are glitches. E.g. whenever the counter h counts up a varying number of bits change. E.g. when counting from 7 to 8 four bits change. Bits 0 to 2 change from 1 to 0 and bit 3 changes from 0 to 1 (7=0111 becomes 8=1000). Now it's impossible to change all those bits exactly at the same time. When e.g. bit 1 switches some picoseconds early then there's a very short moment where the counter is 0101 while switching from 0111 to 1000. And in this very short moment the combinatorics used to generate the sync signal may return a false value. The result is a short spike in the sync signal when there shouldn't be one.

One solution is e.g. to make sync_h and sync_v registers and latch them while the counters are stable.

sorgelig, 2016-01-08:
Till, does your VGA monitor support RGBS mode? I don't think so, since composite sync is for TV only. I think, for VGA monitor you need to change code to output HV sync.
Does your TV support NTSC video? Because native video for NES is NTSC.

We need someone who can test the core with TV supporting NTSC.
I've checked composite sync from this core with oscilloscope and it looks like correct NTSC signal. 63.5us per line, 261 lines per field.

sorgelig, 2016-01-08:
Oh, good thing i discovered while messing with my scalers.
My main scaler is LKV362A which provides good result with upscaling to 1280x720p and converting frames from 50Hz/60Hz to 60Hz. And output it to HDMI. All cores providing PAL or NTSC looks beautiful with this scaler. 
Because this scaler also has HDMI input, i just though, why don't connect my cheap noname MS9282 VGA->HDMI converter and reduce amount of cable reconnections. I've waned to use it for cores providing correct VGA signals (not scandoubled 50Hz) like NES.
And to my big surprise these chain connected converters start to work with scandoubler output! Now i can see video from all cores using scandoubler and output 50Hz preudo VGA output.

So, for those seeking for cheap solution this is one of them. It's really cheap.
The only inconvenience i see, is to press button on LKV362A to switch between SCART and HDMI(connected to VGA converter). 
All input cables like VGA and Audio can be Y-spitted to eliminate cable re-connections.

Till, i can provide more specific info, if you would like to put it into Wiki.

harbaum, 2016-01-08:
> Till, does your VGA monitor support RGBS mode? I don't think so, 
I am absolutely sure, my screen supports RGBS @ 15khz. I have been testing all other cores on it.

For the records: The LCD able to display RGBS@15khz is a EIZO FlexScan L568

sorgelig, 2016-01-08:
My SCART upscaler works fine with NES core. Unfortunately my TV has only YPbPr and HDMI inputs, so i cannot test it on TV. 
It's strange that topicstarter didn't provide test results yet.

pcfreak324, 2016-01-13:
Thank you sorgelig for your update on the NES core. It can confirm it working on my CRT TV. 

Unfortunately, the image is shifted a bit to the left (like with all my consoles in RGB mode). Would it be possible to add an ini setting for changing the horizontal shift of the image?

When using a keyboard, do I have to press key 1 or 2 first to set which joystick I want to emulate with my keyboard?

sorgelig, 2016-01-13:
Currently, there is no way to pass ini settings to core. I think IO Firmware needs extended API to support this functionality.
Since you have this problem with all consoles, then probably it's better to adjust your TV settings. May through some service menu.

MIST supports some USB joysticks directly. Check the Wiki page.

harbaum, 2016-01-13:
The OSD verilog component has access to the video signals and is controlled by the IO controller. It would probably be possible to extend it in a way that it could add and offset to the sync signals. I'll try that.

Mist should support the vast majority of USB joysticks directly.

pcfreak324, 2016-01-14:
Till: Thank you for your efforts. The possibility to add an offset to the sync signal would be a great addition.

Considering the use of joysticks:

Currently I am using an USB joystick. I bought the ones that you recommended onf the Wiki page. The Retrolink Gamecube controller, although a cheap knockoff, works surprinsingly well. The button mapping fits perfectly. I also tried the Buffalo SNES controller but the button mapping for the NES core is a bit uncomfortable. The NES A and B buttons are mapped to SNES A and B. A mapping to SNES B and Y would be more appropriate. This leads to another request: Would it be possible to add support for the button mapping function from mist.ini to the NES core?

Is there an increase in input lag when I use an USB joystick instead of a USB keyboard or a 9-pol joystick? On a PC with Mame there is about one frame of additional input lag when using an USB joystick. Mame uses directinput for USB joysticks and rawinput for USB keyboards which is faster. Therefore, it is recommended to use a joystick with a keyboard encoder which acts like a keyboard (every joystick button is internally mapped to a key).

sorgelig, my joystick with USB keyboard encoder (don't know the brand since I bought it pre-build from eBay) is not detected as a keyboard with the NES core. My cheap Acer keyboard works however. Do you have a special routine that checks if a real keyboard is present and that might prevent my joystick from working? As for the keyboard to joystick mapping, I recommend setting it to the default Mame button mapping for the biggest compatibility with arcade sticks.

Newsdee, 2016-01-14:
There are actually two mappings that you can do with MIST.INI:

First,  a USB mapping from the physical buttons of the controller to a USB
"button1", "button2", etc. These then use a default internal mapping passed
to cores.

Then you can remap the default assignment into one or several keys. In the
wiki example we map the Y button to "up" (for games that do not use a jump
button), but you can switch it to B.

Look at this page for all the gritty details:
https://github.com/mist-devel/mist-board/wiki/USBJoystickMapping

pcfreak324, 2016-01-14:
The mist.ini mapping is not supported by all cores, for example it works with the Amiga core but not with the NES core.

Newsdee, 2016-01-14:
The mapping happens inside the firmware before gamepad data is sent to the
cores. It should be generic, at least last I checked.

How exactly do you experience the problem? Are you using a mapping or
relying on the default for the iBuffalo?

sorgelig, 2016-01-14:
@pcfreak324 
Input devices like keyboard, joystick, mouse are handled completely by IO firmware. FPGA core just gets button press notifications. 
As stated in Wiki, keyboards should support boot mode (the mode you use in PC BIOS, for example) in order to work with MIST. Probably your USB keyboard encoder doesn't support this mode, thus not supported by IO firmware.

harbaum, 2016-01-14:
I doubt there are USB keyboards that don't support boot mode. This was invented to be able to control a PC BIOS without having the BIOS to implement a fill blown HID report descriptor parser. All keyboards i've seen so far support this.

I think @pcfreak324 is just unhappy with the button mapping. This can be adjusted as @Newsdee describes inside the I controller before the joystick events reach the core at all. So this is totally transparent to the core. But this is rather off-topic and should perhaps not be discussed in all detail in a completely unrelated big report ...

sorgelig, 2016-01-14:
The issue (15Khz mode) considered as solved. Reply here is still have issues with 15KHz mode, or open another issue with appropriate topic.

amigrave, 2016-01-31:
OMG !!! I can't believe this !

I just received my VGA to RGB cable, updated my mist nes core, added a `mist.ini` with `scandoubler_disable=1` and then... [IT WORKS !!!](http://amigrave.com/upload/posts/mist-vs-nes/mist-kirby.jpg)

@sorgelig Can you tell me if the necessary changes was "only" this commit e541d27dbe5235f31c5858ce54e340c23ae5a8a4 ?

sorgelig, 2016-02-01:
should be https://github.com/mist-devel/mist-board/commit/3200ed2cb5706150104e98d994eb65dc6c406572 as well.
Just check all my [NES] commits in repository.

juri74, 2016-02-05:
Hello, sorry if i bring this up! but i can't actually find latest fixed core (commit e541d27 or 3200ed2) 
could you help me please? i checked all your commits but i can't find any [NES] ..
oh btw i'm impressed by your BK0011M core!

sorgelig, 2016-02-05:
here: https://github.com/mist-devel/mist-binaries/tree/master/cores/nes

juri74, 2016-02-05:
thank you sorgelig.. 
so tested and unfortunately i have to report a malfunction :'(
i used the same 15khz rgb cable for amiga,atari st, zxspectrum 128 and c64 (they work perfectly and c64 in ntsc mode display a stable picture)
the screen is showed an in it's complessity is good, but it continously move left and right of (i think) 8 pixels, timed by around half seconds..
the last bottom bar (again i think 8 pixels) is drawed almost correctly but, when the screen is on the left it miss some pixels and show noise. i can make a vid if it helps..
thanks

juri74, 2016-02-05:
https://www.youtube.com/watch?v=f8TECc2YDqU

MmmPT, 2016-02-05:
I quit trying....Sometimes works with some tvs or monitors ....sometimes it
does not. I just gave up

sorgelig, 2016-02-05:
@juri74 
Does it happen with every game or specific?
It does work ok with my SCART->HDMI converter. I will see what can i do.

juri74, 2016-02-05:
hello it happen on all games i tested, and it happen even at core start, before you select a game, the osd is affected too

juri74, 2016-03-19:
hello, any nice updates? :)

[7 more replies]

closed 2020-06-16 by gyurco