mist-devel/mist-board #18: C64 SID problems.

sorgelig, 2016-01-14:
The SID code in C64 core doesn't look accurate. Sound in some tunes looks strange or messed in some cases. 
For example, Intro with dancing girl (again dancing girl, haha!) in Hollywood Poker Pro is very messed.
And even tunes primarily considered OK, sound strange. Turrican II Title sound is too jerky...

One problem in SID code is noise generator. The second is total filter absence in code.

I've found another, unfortunately unfinished, SID FPGA project in [fpgaarcade](http://www.fpgaarcade.com/punbb/viewtopic.php?id=256) forum. That project is based on chip decaping according to author. I've made it work with C64 core, it also has issues, but what is working, looks correct. It seems like one channel is not working, and i didn't find the source of problem. But noise does look correct. It also has filters implementation, but according to author the filter part hasn't finished yet. I've imported noise generator to existing SID code from fpgaarcade project. Noise is better, but not enough to be acceptable.

Does anyone now other FPGA SID implementations? Or may be someone can help to fix the problems by mixing these 2 SID implementations?

Newsdee, 2016-01-14:
The SID of the TC64 core sounds exactly like the real thing to my ears.
While I'm not a musician the difference is pretty clear, for both SID
models.

Unfortunately it isn't open-source...

sorgelig, 2016-01-14:
Not open source is useless here.
And by the way, i've read on some German forum about TC64's SID, that is has many unfinished or non-working parts in code. Some tunes can sound ok, but some are not.

Did you try Hollywood Poker Pro Intro on TC64? https://www.youtube.com/watch?v=mn4pGa5nIR4

sorgelig, 2016-01-16:
May be i will tell crazy thought, but what if.... make some core with high enough CPU clock and run VICE inside it? Because VICE looks very polished and mature comparing to FPGA64. And it looks like nobody works on code anymore.
Perhaps can make CPU core for SID emulation, but reSID uses large tables and i'm afraid won't fit into FPGA RAM since FPGA64 already occupies 2/3 of available RAM.

harbaum, 2016-01-16:
Does vice run fast and accurate enough on the amiga 500 or the Atari ST? That's about the fastest we have on the mist today.

It might be possible but instead of writing a whole c64 in hdl you'd write a complete system able to run the vice emulator. I'd assume that might even be more work.

harbaum, 2016-01-16:
There is another c64 core which is also named fpga64: http://fpga64.blogspot.de 

I once was in contact with him but haven't received in answer for over a year. That core is lacking floppy support which we now have. It would imho make much more sense to get in touch with him again.

sorgelig, 2016-01-16:
In the mean time i've got more ideas. 
What about this one in FPGA? http://www.swinkels.tvtom.pl/swinsid/
It seems provides full replacement for SID in simple AVR chip. 
I've checked opencores and it has couple AVR models. Atmega8515 has only 8kb ROM and 512b RAM.

sorgelig, 2016-01-16:
About that FPGA64 project: i've found the sources used in project here https://svn.pacedev.net/repos/pace/sw/src/component/sound/sid/  It uses the same SID model as MIST.
Found this info from following video: https://www.youtube.com/watch?v=4bhIrPKqFII

harbaum, 2016-01-16:
Ah, indeed. Too bad. But the rest seems pretty cool and may be worth a look, anyway.

robinsonb5, 2016-01-16:
On the subject of systems fast enough to run VICE, the f32c MIPS/RISCV core is interesting - in MIPS configuration the CPU is significantly smaller than the TG68 but it's ridiculously fast - it can turn in about 180DMIPS @ 100MHz.

sorgelig, 2016-01-16:
Hmm. Tried to register on fpgaarcade forum but got rejection (and it seems manual admin approval). It seems not everyone allowed to be there.
Just wanted to ask in SID topic about status...

rkrajnc, 2016-01-16:
Be careful with code from fpgaarcade.com - AFAIKat least some code there is only allowed to be used on the fpga arcade board, so it's not really open-source, or at least not GPL - compatible.

As for running VICE on MiST, my guess is that this emulator is way too complex, so it would be really hard to run it full speed on any CPU architecture.

BTW, I have a Pentium 200 system with 133MHz SDRAM, which I guess is surely above the maximum performance one could get from a Cyclone III FPGA, I'll try to run Vice on it and see if it works normally.

sorgelig, 2016-01-16:
SID code doesn't limit usage on fpgaarcade board only. No GPL mention but free to use and redistribute. Anyway, the code is also not working well. 
Actually both original SID code and from fpgarcade have problem with noise simulation or something relative to it. 
It looks like AVR simulator is the best way to go. The only problem, is source code absence of latest nano-SwinSID firmware for ATmega88. They have source code for older version which is much older. 
Also, AVR module is for ATmega103 and need to be tweaked for Atmega88 (because source code of firmware is absent).
I'm not sure if my experience will be enough to handle this implementation.

fpgaaracde, 2016-01-17:
Hi.
rkrajnc - There are some cores by other authors under different licensing. Some of the board specific libraries are under "only use here" license as they are written specifically for the replay board. All my stuff, including the cores, is fully open source. If in doubt read the license or ask a question.
This will be further relaxed shortly.

sorgelig, I have no idea why you got rejected, please email and I'll fix it. The forum is getting a huge amount of spammer registrations and this was a necessary step. I posted some comments apologizing for this and asking anybody who did not receive a mail to contact me.

I intent to look at the SID filters again shortly.
/Mike

rkrajnc, 2016-01-17:
Mike: OK.

sorgelig: maybe it is a better idea to check the SwinSID code and see if we can fix the current SID implementation, instead of replacing it with a ATMega core and related code - it just seems simpler to do.

I can check the noise channel implementation - I guess this is a simple LFSR pseudo-random noise generator, so all that is required is figuring out what kind of polynomial is used, and the initial state.

sorgelig, 2016-01-18:
@fpgaaracde  Mike, glad to see you here! My mailbox name is very suspicious, right :) It's because i've made it long time ago just for registering on boards and eventually receive the spam. When i registered on github i didn't expect to participate in development, but it turned out i'm in. My mailbox is pour.garbage_at_gmail.com I've tried to mail to admin of board without success.

That's good, you will continue your SID library. Additional to Noise, there is something strange with channels. Hollywood Poker Pro(HPP) Intro looks like one channel is missing. I have your SID library integrated into C64 code. If you want, i can send you the code for testing. 
It seems HPP Intro uses some special tricks in music and it's good for testing.

@rkrajnc 
yes, it's LFSR, and the code is seems correct, but it looks like not working. In original code the beginning of HPP Intro is always noisy, in Mike's library the noise sounds not like noise. At the same time in hacker's intro before original intro i hear good drums with noise.

I've learned how to implement true RNG using LFSR with cell ring oscillator while working on Menu core. So, may be it can be used in SID.

fpgaaracde, 2016-01-18:
@Sorgelig,
The LSFR noise generator, taps and mux is correct (as far as we can 
tell) from the layout. There are some additional complexity with 
feedback when multiple waveforms are turned on at the same time which 
isn't quite right.
Something else is going wrong, almost certainly in the "analog" stage. 
Please send any changes you make back so I can merge into our SVN.

Which mail address did you sent to, I received nothing (which is 
worrying). Please register again and I'll check the logs.
/Mike

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

sorgelig, 2016-01-18:
@fpgaaracde 
I've just registered of forum finally. I've used other e-mail.
My changes aren't meaningful for SID code itself. I've just changed the code so, interface works at 32MHz while internal modules like voices, filters at original 1MHz frequency. I can give you the whole C64 source code with your SID module integrated, if it can help you to debug. Or may be just to check if i did it correctly. 
Too bad, I cannot simply publish the source code on the github.

fpgaaracde, 2016-01-18:
Hi,

@fpgaaracde <https://github.com/fpgaaracde>
>
> I've just registered of forum finally. I've used other e-mail.
>
Great.
> My changes aren't meaningful for SID code itself. I've just changed
> the code so, interface works at 32MHz while internal modules like
> voices, filters at original 1MHz frequency. I can give you the [...]
You can as long as the header is in place. Take the files as is, and add 
a line at the top, something like

"modifed by XXX, original at svn.fpgaarcade.com"

and your changes. I'll merge back any fixes.
I do this when I take GPL code into my public SVN etc, although I double 
check with the author.

Why wouldn't clocking the code at 32MHz with a 1M clock enable work with 
the original code?

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

sorgelig, 2016-01-18:
Why don't you want to upload repository to github?

fpgaaracde, 2016-01-18:
This has been discussed on the forum. Several reasons, the main one is 
we have dedicated servers running SVN that we control so there is no need.
The other is there is some code which is not public yet, either because 
we haven't clarified the license or it cannot be released (some of the 
ASIC stuff). Mirroring between dev and public SVN works for now for me.

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus