mist-devel/mist-board #19: about resolution in video.v

thecrazyboy, 2016-01-15:
Hello everyone
first, I don't know if it's the right place to ask questions about the HDL code.

I have checked the video.v, 
I think your idea it is put orignal 256*240 (mode = 1), 512*480 (mode=0,smoothing=0)and 512*480 with hq2x optimzation(mode=0,smoothing=1) into one module which is very good.
but it raises some questions to me,
1. all above 3 display methods are 640x480 resolution? 
2. take your code:
wire hpicture  = (h < 512);        //this does not make sense, why not     hpicture  = (h < (512>>mode));
wire hend = (h == 681);            //again ,why not     hend = (h == (681>>mode) )
wire vpicture = (v < (480 >> mode));    
wire vend = (v == (523 >> mode));       

assign       VGA_VS = mode ? 1'b1 : ~sync_v;//when mode =1, VGA_VS is always high???

have you test it when mode = 1?

3. have you test nes on modelsim?
I have tried and failed ,because many signals are unkown (x state) during simuation cause cpu stop at some instruction.
and I do some modification by initialize signals in initial statement or during reset , but I can‘t fix it all because i am a newbie.

thecrazyboy, 2016-01-15:
about 2rd question, I checked issue list "15Khz support for NES core", maybe  mode =1 is for TV not VGA?
b

sorgelig, 2016-01-15:
As far as i understand the questions are about NES core. There are many cores in this repository, so be more specific when you ask. 
The NES core originally was developed not for MIST board. So, the question #2 about quality of code you need to address to original NES author who is not in MIST community, i believe.
The core just had been ported to MIST without code refactoring by several reasons. If you want to improve the code then you are welcome to do it. Just make sure you don't brake it ;)

about #1: 
For some reason, the master clock of pixels is twice higher than original. Hence double sizes in h counter (2 ticks per pixel). Since in VGA mode frequency is twice higher, the total h size is the same for TV and VGA.
But in TV mode there is half amount of lines compared to VGA, thus vertical values are adjusted to mode.
Total resolution in TV mode (mode=1) is 342x262, in VGA (mode=0) is 682x524.

In TV mode composite sync (H+V) goes to VGA_HS, while on VGA_VS is always 1 for AV-input switch in SCART. Check the Wiki page about VGA->SCART cable.

harbaum, 2016-01-15:
We now have people asking for a second NES core ... I wish someone would start working on a SNES core instead :-) I'd be the first one to offset help as I'd like to show my kids how Mario Kart looked when it all started.

harbaum, 2016-01-19:
Please close this issue if your question has been answered.

closed 2016-01-28 by harbaum