mist-devel/mist-board #21: Nice smoke test for the C= 64 core

ArKay74, 2016-03-28:
This demo seems to work only on real hardware and in micro64. VICE, Emu64 and MiST fail.

http://csdb.dk/release/?id=146723

harbaum, 2016-03-31:
Yes, the C64 isn't perfect. It's license isn't 100% clear. We got permission from the origjnal author to publish cores based on it. But we cannot simply share the source and thus it doesn't get much attention I am afraid.

ArKay74, 2016-03-31:
I have the feeling that there isn't much missing so that's really sad. 
How cool it would be if it would run this demo
which has received quite some attention. The guy who tested 37 games on 
YT only had problems with a few of them.
So we can only hope for the Mega65 to get it right then.

harbaum, 2016-03-31:
Actually the last 10% of the games and demos require 90% of the coding effort.

These are the games and demos that rely on perfect timing which oft means that you need to rewrite everything.

juri74, 2016-06-04:
hello, where is the RESTORE key mapped? can't find it..

gyurco, 2020-04-27:
I think this demo works well now.

closed 2020-04-27 by gyurco