mist-devel/mist-board #22: multicomp really ported to mist?

Creep69, 2016-03-31:
In the fpga projects section of this wiki it reads:

Multicomp 	Link 	yes 	Has been ported to the MIST

Has this really been ported to MiST, already? If so, where can I find this port? I built the multicomp Z80 CP/M system using the same Chinese Cyclone2 board and was considering to give it a try to port it to MiST, myself. But only if it does not exist, already.

Rene

harbaum, 2016-03-31:
If i remember correctly the guy behind http://ws0.org/ did it a long time ago. Maybe you contact him.

Creep69, 2016-03-31:
Thanks Till, I will try this. Since the multicomp project is so well 
documented, it should find its way to MiST and is a good project to 
start with.

Rene

Creep69, 2016-04-03:
Found the project here:

http://ws0.org/building-your-own-custom-computer-with-the-mist-fpga-board-part-1/

From scratch it works to build the 4KB Z80 Basic computer. Up to 32KB RAM was possible with FPGA-SRAM, for more we need to utilize the SDRAM. When this is done and accessing SD-Card works, too, it should be possible to built the whole CP/M computer for MiST. If I succeed, I will give notice.
Rene

closed 2016-04-03 by Creep69

wsoltys, 2016-04-03:
Yeah, its quite some time that I build that. Please go ahead and try to extend it. Using sdram should be possible with Till's wrapper. SD card might be a little trickier.

harbaum, 2016-04-03:
The sdram of the mist is a standard one. Any core supporting sdram should easily be ported. The sd card needs a wrapper as the sd card is not connected directly to the fpga on the mist.

For both, the sdram as well as the sd card are examples available in the mist tutorials:

Https://github.com/mist-devel/mist-board/tree/master/tutorials/soc

Creep69, 2016-04-04:
Hi both, thanks for the hints. It's my first FPGA project and I'm already working with the tutorials. I'm confident to get the SDRAM connection working. Regarding SD card, even to use it to load/store images might be enough. The CP/M System Alteration Guide describes how to port it to almost any architecture. Since Grant Searle used a serial terminal to transfer the system files to the Z80 system and then to the SD card, I have to do some modifications, anyway. Good project for learning!