mist-devel/mist-board #24: Split the sources

sorgelig, 2016-06-03:
@harbaum,
Could you consider to split the sources of cores into separate repositories?
It will be more easier to provide the support because developers can work on specific cores without forking into separate repo. 
Also, it would give a chance to prune non-used sources. For example ZX and minimig cores are no longer used in this repository and may mislead new developers. 
Wiki already provides links for repositories for each core, so it can be used as a central hub to gather all used repositories.

harbaum, 2016-06-03:
Why not? This could be done while someone is working on a core.

sleary78, 2016-06-03:
I'm happy with the current structure BUT! I'm happy to prune dead cores. If there are cores that arent getting love let me know and I'll pick them up.

Newsdee, 2016-06-04:
Hi Stephen,

If you mean improving some older cores, may I suggest the NES and Gameboy
cores?

They could use a bit of work to improve compatibility (specifically,
support more memory mappers); and ideally would need to backup "Save RAM"
to the SD card so one can play longer games (eg.  Zelda).

Thanks,
Gerard.

gyurco, 2020-05-29:
As I'm usually hacking on several cores at a time, my local repo has a lot of "garbage" (unfinished or experimental code). I think it would be easier to handle with separate repos for the cores. If it's OK, I can generate FPGA64, C16, BBC-micro, Archimedes repos with this method:
https://help.github.com/en/github/using-git/splitting-a-subfolder-out-into-a-new-repository
This one keeps the commit history, so regression-hunting via bisecting will still work.

harbaum, 2020-05-29:
Yes, of course feel free to cut things. When I started I didn't expect much more than the Atari st and  the Minimig. But that prediction was slightly wrong ....

gyurco, 2020-05-29:
MiST is like a good wine, it just gets better with age :)

gyurco, 2020-06-16:
It's done.

closed 2020-06-16 by gyurco