mist-devel/mist-board #25: Why R21 used?

sorgelig, 2016-06-08:
While debugging SAA1099 module and pulling my hair from everywhere while trying to fix stereo sound, I've discovered MiST having very noticeable audio channel cross-talking.
An then I've discovered resistor R21 on the board mixing LR together. 
Why?
I'm not 100% sure if it will fix the problem when i remove this resistor, but i have no idea what is the purpose of this resistor.

sorgelig, 2016-06-08:
Hurrrraaa! I've fixed SAA1099 sound by removing R21!

rkrajnc, 2016-06-08:
The resistor is (probably) a leftover of the minimig design, and is supposed to mix the left and right channels somewhat. This is Amiga specific, as each of the four Amiga sound channels is hard-mapped to a left or right speaker, and it produces a hard stereo separation.

It's a shame that the resistor is there, as the mixing could easily be done inside the FPGA, if required.

harbaum, 2016-06-08:
Indeed, it's been taken from the minimig.

harbaum, 2016-06-08:
When i designed the board even in my wildest dreams I would never have expected that it to be running more then the minimig and the atari st cores.

sorgelig, 2016-06-08:
Probably it's better to remove this resistor from future produced boards.
SAA1099 has special feature "stereo envelope" where one channel has rising volume while another channel mirrored (decreasing volume). R21 destroys this feature and some audio FX play incorrectly.

EricGus, 2016-06-08:
Any consequences of removing it in regards to the Atari/Amiga cores? .. I'm just trying to understand why it was put there on the original minimig to start with (i.e. what problem it was solving there)

harbaum, 2016-06-08:
It doesn't really solve a problem. The normal atari st sound will not suffer as it's monophonic. The same is true for most other cores. Only on machines with stereo output will removing the resistor result in a stricter stereo separation. Imho the Amiga core will suffer a little bit as the real Amiga had the same resistor if I remember correctly. This can be re-implemented inside the core as Rok correctly points out.

Imho you can just remove it. Give it a try. It won't hurt  :-)

sorgelig, 2016-06-08:
Not sure what you meant, but It does SOLVE the problem.
For mono cores there won't be any changes as both channels repeat each other. For Amiga, @rkrajnc will add mixing inside the FPGA since he is aware already. Although, it's matter of preference how hard stereo you want. 
BK0011M and ZX cores have 3 audio channels where one of them has been split between 2 channels, so for these cores removing R21 virtually won't affect at all.

For SAM Coupe core this resistor should be removed as it affects the sound due to nature of SAA1099 sound.
IMHO it's better to remove R21. If any core will have to strong stereo separation, then small and easy fix can be added into FPGA code. Other way you cannot fix if R21 is soldered.

robinsonb5, 2016-06-08:
Real Amiga's don't have any inbuilt stereo mixing.  The resistor's only there, I think, to reduce stereo separation a little for headphone listening - total separation is quite uncomfortable through headphones.

EricGus, 2016-06-08:
@sorgelig .. ok thanks this is exactly what I was hoping it was used for.. ok guess its time to bust out my reading glasses and soldering iron.

rrmagro, 2016-06-11:
This whould be nice to add to the wiki so people would be informed, maybe a new section, something like mods or tweaks or even hacks.