mist-devel/mist-board #28: Plus_Too core: which .DSK and .IMG format?

sblendorio, 2016-07-18:
Hi, I downloaded System6 .DSK files from here:
https://sites.google.com/site/minivmacapplicationsv6/systems-os
(specifically: https://sites.google.com/site/minivmacapplicationsv6/systems-os/System608.zip?attredirects=0&d=1)

...and a hard disk images from here:
http://digiex.net/downloads/download-center-2-0/applications/8541-mac-os-6-games-apps-emulator.html

None of them seems to work with plus_too core. What is the correct disk format? Is it possibile to convert them in a plus_too core compatible format?

harbaum, 2016-07-18:
There's a description incl. links to matching dsk files in the repository:
https://github.com/mist-devel/mist-binaries/tree/master/cores/plus_too
Please check those first.

Harddisk images seem to have pretty emulator dependent formats and thus cannot simply be exchanged with the MIST. The readme mentioned above also explains how to create a hdd image.

The HDD images needs to be in a format a real Mac could work with if e.g. using some SD/SCSI adaptor. The same goes for Floppy images. They also need to be in a format a real mac would expect. Some emulators can a a lot of magic in the background to cope with alien formats. This magic isn't present with real hardware like the MIST or a real Mac.

Newsdee, 2016-07-18:
I use HFVExplorer to transfer files from Mac disk images to the one compatible with the MiST. It can also create new floppy images that are compatible by cloning an existing one and removing contents.
Give it a try: http://www.emaculation.com/doku.php/hfvexplorer

sblendorio, 2016-07-18:
Thank you very much... I should have read the "F-----Manual" before :)

EricGus, 2016-07-18:
You can also try Basilisk II .. I believe I used that to transfer hard disk images (I'm using a real Mac but the current OS X doesn't support those old types anymore)

harbaum, 2016-08-11:
Seems to be solved ...

closed 2016-08-11 by harbaum