mist-devel/mist-board #29: Commodore 64 core dont load TAPs images

Julitium, 2016-08-10:
Hello, I don't know if I should put this here...
 it would be possible to implement in the core of commodore 64 load from TAP or T64 load images?

Thanks.

harbaum, 2016-08-10:
It's sure possible. But it's some work as you'd make the core being able to receive, decode and replay a TAP file into the C64 itself. And it would be as slow as a real tape on a real c64 as we cannot do all those nasty tricks emulators are able to do. 

But what for? There's already the direct PRG injection and the floppy disk. Why a third way to load files?

Julitium, 2016-08-11:
Hi. I also think it is possible. I do not think it necessary to make major modifications. 
loading speed as a real c64 is desirable, because it would make sense to do a load from tape to an unreal speed, I do not think a load from tape is so exasperating bearing in mind that the average time is about 3 minutes
In other systems such as spectrum, Amstrad and other... Load from tape is more slow and boring.
about the whys .....
loading from tape C64 is the essence of the times when we used the system back in the 80's ....
multicolor screen presentation, hypnotizing Borders, musics (and minigames) "in-load" etc.
for instance, I personally have a C64, C64c, C128 and 1541 but always use the Tapuino as a way to load games. I have the taste of the loading of the 80's without the inherent problems of magnetic media (load failures, broken tape, degaussing etc).
really it costs me a lot to decide whether to acquire a unit MiST just this question.
load from disk is fine but do not think there entire catalog that appeared on tape (at least directly) ....
injection by PRG do not think in my case comes to Use it.
For all these reasons I request that is includes loading from
TAPs or T64 (preferably TAPs).

Thank you very much.

harbaum, 2016-08-11:
I thought you were asking someone else (me) to implement that. If you actually think it's only a minor effort for you to do it then of course, please go ahead. Any contribution is welcome.

It's just that I personally don't see sufficient benefit to spend several days implementing this. I am afraid it's less trivial than you expect it t be.

Newsdee, 2016-08-11:
You can use DirMaster to extract PRG from T64 files:

http://style64.org/dirmaster

harbaum, 2016-08-11:
Ah ... we are talking about the C64. I confused it with the C16/Plus4.

Julitium, 2016-08-12:
Hi. you're right ... I was asking that the person who is working on the C64 core could do this. If I said no major changes is that because the C64 receives data from Datasette in digital (no analog audio) ....
I gave my opinion because I assemble, repair and fix all these machines 8 & 16 bits, but I have no specific knowledge of VHDL or Verilog Since I have very young daughters and I did not have much time to study (although it is the first thing I do when I have time ) .
in short, if anyone is encouraged to do this (no hurry) , I would be very grateful .

Thank you very much.

P.D : A confusion happens anyone, and thanks for the extractor T64 PRG !!

EricGus, 2016-08-12:
Actually I have been looking into repurposing those tx/rx (MiST midi serial connector) for use as a c64 user port for modem communications (with tcpser on a rasp pi, much like the Amiga and ST cores can do now). I don't even have any idea how complicated trying to do a cassette interface would be due to the digital nature of the commodore cassette drive.

pooky2, 2016-09-22:
Hello harbaum, the reason I hope to load TAP is because the load of some games had music and is a feature that did not have any other 8 bit computer , for example https://www.youtube.com/watch?v=-X2zNe3YFNM

harbaum, 2016-09-23:
Ok. It's still a significant amount of work. We'll need to find someone skilled enough **and** interested enough in TAP usage to implement this.

EricGus, 2016-09-23:
Implementing the mist midi serial lines as C64 serial on user port would be easier then trying to do tape. For serial It's just connecting the mist RX TX lines to the CIA.  (Like how rs232 is done on the Amiga and ST cores)

Julitium, 2016-10-11:
Hello.
I can see that not am the only interested in the load from images of tapes. pooky2 is right, the commodore 64 has a load from cassette that does not have any other 8-bit machine. the load from floppy not are excessively more fast, but it lost the flavor original with the loads of tapes (graphics and sound SID).
EricGus, your input is very interesting and can be used for this purpose... the problem would be that is that in addition to Rx and Tx is the other two signals (Motor and Sense) which are controlled by the processor. In addition these 2 signals not are TTL.
Harbaum, know that not are many interested (or that may seem)... not is has proposed the addition of this option of tape. 
harbaum, I know that we are not many interested (or that may seem)... no it is proposed the addition of this option of tape, but I am sure that the owners of commodore 64 would appreciate it.
I still think that an implementation should be more faithful to the real system.
There is no hurry, but in the long term, the implementation of commodore would be at a level very high and varied (and accurate).

thanks.

EricGus, 2016-10-11:
Actually Julitium I was more advocating using the MiST RX/TX lines to implement the C64 USERPORT to allow serial / modem (via something like tcpser) communications and NOT to be repurposed as a cassette tape input (mostly due to the complexities of how commodore implemented tape control). setting up serial "aka rs232" on the c64 would just be a matter of connecting a few lines from the CIA of the c64 core to the TX/RX MiST pins.

RS232-TTL               C64 User Port Pin
GND                           A & N
TXD                            M
RXD                            B & C
VCC                            2

Julitium, 2016-12-03:
Hello,
I would like to reconsider the possibility of loading from .tap.
Other fpga systems are implementing them successfully and causing a great deal of interest in their user group.
Thank you .
greeting

EricGus, 2016-12-03:
Are you just talking now about loading a .tap file from the F12 menu? .. that would be significantly easier to implement then trying to wire up an actual cassette player to the c64 core. That said there is no one actively developing or working on improvements or adding new features (like enabling disk writes or 1541 drive compatibility) to the c64 core that I know of (aside from a few much needed fixes that Sorgelig did to address NTSC display and improvement to the SID and speed up .d64 disk loading) .. the author of the fpga64 core "closed" the source after inclusion of proprietary code for the turbo chameleon and no longer publishes his updates.

pooky2, 2016-12-03:
Sure, I meant uploading files from the F12 menu. The idea is to simulate
the load of the tape so that MIST plays the load music. I do not need to
connect a real cassette to the MIST board.

renaudhelias, 2016-12-05:
Could be nice also for me : reading .CRT and .DSK in a future version of CoreAmstrad.
Could be nice to see files extension (in OSD files list) in case you enable two types of file entry.

Perhaps something like that :
CONF_STRING : string := "AMSTRAD;DSK;O1,Brand name,Schneider,Amstrad;O2,Drive,A,B";
CONF_STRING : string := "AMSTRAD;**DSK,CRT**;O1,Brand name,Schneider,Amstrad;O2,Drive,A,B";

Julitium, 2016-12-05:
Yes, pooky2 is right.... it's about implementing the .tap load from the OSD (F12), otherwise it would be complex and tedious for most users. There is some periphery for real C64 like Tapuino.
In the real system this is a problem (you can not use a normal player) due to the digital nature of the data, but in an implementation supported by ARM this should not be complicated (if you are already programmed the decoder to handle .tap) .
The load from tape in C64 is a rich experience in senses and emotion (Colors, SID, Minigames, etc).
Thank you.
regards

EricGus, 2016-12-06:
As I mentioned previously there is no one actively developing the c64 core.

lroby74, 2016-12-28:
I would like to load .TAP (from F12 menù) too but is not possible to take last old open C64 core source and create a fork? Like in Linux?

lroby74, 2019-04-18:
The TAP loading from OSD F12 menù in reality now! And now can load and save from and to a real tape recorder on Mistica too!!

Julitium, 2019-04-19:
Sure, my friend!

lroby74, 2019-04-27:
Is possible to close this issue?

Julitium, 2019-04-27:
Of course my friend!!!....
My dream is now a reality. 

Thanks!!!

closed 2019-04-27 by Julitium