mist-devel/mist-board #40: BBC Micro core no OSD with 15khz RGB

squidgydude, 2016-11-14:
The OSD works fine in VGA but with 15khz RGB it does not appear.

squidgydude, 2017-02-08:
Additionally the colours look dark and washed out on RGB. Perhaps running at 0-255 instead of 16-235?

squidgydude, 2017-02-20:
I've fixed the OSD not appearing in RGB 15khz and fixed the washed out colours in RGB 15khz (VGA wasn't affected). Binary posted [here](http://www.atari-forum.com/viewtopic.php?f=101&t=27906&start=25#p311696). 

Here's the patches:

`$ diff -Nur bbc_mist_top.v bbc_mist_top.v.orig
--- bbc_mist_top.v      2015-10-09 08:53:27.000000000 +0100
+++ bbc_mist_top.v.orig 2017-02-12 17:56:54.026881300 +0000
@@ -109,7 +109,7 @@
 
 // without scan doubler only half the pixel clock is used
 reg clk_12m;
-always @(posedge clk_24m)
+always @(clk_24m)
        clk_12m <= !clk_12m;
 
 wire clk_osd = scandoubler_disable?clk_12m:clk_24m;
@@ -133,9 +133,7 @@
    .green_out  ( VGA_G        ),
    .blue_out   ( VGA_B        ),
    .hs_out     ( v_hs         ),
-   .vs_out     ( v_vs         ),
-
-   .tv15khz    ( scandoubler_disable )
+   .vs_out     ( v_vs         )
 );
 
 wire v_hs, v_vs;
`
`$ diff -Nur osd.v osd.v.orig
--- osd.v       2017-02-20 10:44:38.963717447 +0000
+++ osd.v.orig  2015-09-30 11:33:46.000000000 +0100
@@ -45,9 +45,7 @@
        output [5:0]    green_out,
        output [5:0]    blue_out,
        output                  hs_out,
-       output                  vs_out,
-
-    input           tv15khz
+       output                  vs_out
 );
 
 parameter OSD_X_OFFSET = 10'd0;
@@ -196,16 +194,9 @@
   osd_byte <= osd_buffer[{osd_vcnt[6:4], osd_hcnt}];
 
 wire [2:0] osd_color = OSD_COLOR;
-
-// RGB 15khz output is different to fix dull colours, VGA not affected.
-// BBC only uses one bit per colour but video engine uses a 2-bit vector.
-wire [5:0] r_in = tv15khz? {5{red_in[0]}}   : {3{red_in}};
-wire [5:0] g_in = tv15khz? {5{green_in[0]}} : {3{green_in}};
-wire [5:0] b_in = tv15khz? {5{blue_in[0]}}  : {3{blue_in}};
-
-assign red_out   = !osd_de? r_in :  {osd_pixel, osd_pixel, osd_color[2], {red_in, red_in[1]}  };
-assign green_out = !osd_de? g_in :  {osd_pixel, osd_pixel, osd_color[1], {green_in, green_in[1]} };
-assign blue_out  = !osd_de? b_in :  {osd_pixel, osd_pixel, osd_color[0], {blue_in, blue_in[1]} };
+assign red_out   = !osd_de? {3{red_in}}     :  {osd_pixel, osd_pixel, osd_color[2], {red_in, red_in[1]}  };
+assign green_out = !osd_de? {3{green_in}}   :  {osd_pixel, osd_pixel, osd_color[1], {green_in, green_in[1]} };
+assign blue_out  = !osd_de? {3{blue_in}}    :  {osd_pixel, osd_pixel, osd_color[0], {blue_in, blue_in[1]} };
 
 assign hs_out = hs_in;
 assign vs_out = vs_in;
`

squidgydude, 2017-02-20:
I've fixed the OSD missing on RGB 15khz and fixed the washed out colours in RGB 15khz (VGA wasn't affected). I've posted the binary [here](http://www.atari-forum.com/viewtopic.php?f=101&t=27906&start=25#p311696). 

Here's the patches:

```$ diff -Nur bbc_mist_top.v bbc_mist_top.v.orig
--- bbc_mist_top.v      2015-10-09 08:53:27.000000000 +0100
+++ bbc_mist_top.v.orig 2017-02-12 17:56:54.026881300 +0000
@@ -109,7 +109,7 @@
 
 // without scan doubler only half the pixel clock is used
 reg clk_12m;
-always @(posedge clk_24m)
+always @(clk_24m)
        clk_12m <= !clk_12m;
 
 wire clk_osd = scandoubler_disable?clk_12m:clk_24m;
@@ -133,9 +133,7 @@
    .green_out  ( VGA_G        ),
    .blue_out   ( VGA_B        ),
    .hs_out     ( v_hs         ),
-   .vs_out     ( v_vs         ),
-
-   .tv15khz    ( scandoubler_disable )
+   .vs_out     ( v_vs         )
 );
 
 wire v_hs, v_vs;
```

```
$ diff -Nur osd.v osd.v.orig
--- osd.v       2017-02-20 10:44:38.963717447 +0000
+++ osd.v.orig  2015-09-30 11:33:46.000000000 +0100
@@ -45,9 +45,7 @@
        output [5:0]    green_out,
        output [5:0]    blue_out,
        output                  hs_out,
-       output                  vs_out,
-
-    input           tv15khz
+       output                  vs_out
 );
 
 parameter OSD_X_OFFSET = 10'd0;
@@ -196,16 +194,9 @@
   osd_byte <= osd_buffer[{osd_vcnt[6:4], osd_hcnt}];
 
 wire [2:0] osd_color = OSD_COLOR;
-
-// RGB 15khz output is different to fix dull colours, VGA not affected.
-// BBC only uses one bit per colour but video engine uses a 2-bit vector.
-wire [5:0] r_in = tv15khz? {5{red_in[0]}}   : {3{red_in}};
-wire [5:0] g_in = tv15khz? {5{green_in[0]}} : {3{green_in}};
-wire [5:0] b_in = tv15khz? {5{blue_in[0]}}  : {3{blue_in}};
-
-assign red_out   = !osd_de? r_in :  {osd_pixel, osd_pixel, osd_color[2], {red_in, red_in[1]}  };
-assign green_out = !osd_de? g_in :  {osd_pixel, osd_pixel, osd_color[1], {green_in, green_in[1]} };
-assign blue_out  = !osd_de? b_in :  {osd_pixel, osd_pixel, osd_color[0], {blue_in, blue_in[1]} };
+assign red_out   = !osd_de? {3{red_in}}     :  {osd_pixel, osd_pixel, osd_color[2], {red_in, red_in[1]}  };
+assign green_out = !osd_de? {3{green_in}}   :  {osd_pixel, osd_pixel, osd_color[1], {green_in, green_in[1]} };
+assign blue_out  = !osd_de? {3{blue_in}}    :  {osd_pixel, osd_pixel, osd_color[0], {blue_in, blue_in[1]} };
 
 assign hs_out = hs_in;
 assign vs_out = vs_in;
```

squidgydude, 2017-02-21:
I've committed the fixes to the repository.

closed 2017-02-21 by squidgydude

harbaum, 2017-02-21:
Excellent,  thanks a lot.