mist-devel/mist-board #41: MiST Color Mode Constrained

Televicious, 2016-11-16:
On the MiST core set to color mode the screen is scaled far too large for some LCD monitors. Maybe there's a way to multiply the VGA video output resolution? All others scale properly except MiST color mode.

harbaum, 2016-11-17:
The video scaler is a simple scan doubler which just doubles every line but keeps all timing close to the original otherwise. Only this allows for smooth and synchronous scrolling and similar effects.

Scaling to a totally different screen size is possible but has a big complexity and requires RAM and FPGA resource we'd better leave for use by the core itself.But the main problem: If you separate the cores video timing from the VGA video timing you'd get the same tearing and stuttering video many emulators have.

Did you try a US/NTSC TOS? Or did you try to use some software like some menu disc intros that allow you to switch to NTSC/60Hz? If that helps then your screen simply doesn't cope with the old 50Hz PAL video the core uses for ST compatibility reasons. Then there's not much we can do about that.

Televicious, 2016-11-17:
Oooo! Didn't know there was an NTSC specific core. Thought maybe there would be software compatibility conflicts like on the C64. I'll definitely give that a go and see if it works. Thank you very much.

harbaum, 2016-11-18:
You are talking about the Atari ST core, right? It's not a special core but the TOS or other software that tells the hardware what video mode to setup. Using an american TOS will result in a 60Hz NTSC video on the same core.

Televicious, 2016-12-28:
I tried the NTSC core with the exact same results. I believe it's just too low of a resolution for a lot of modern LCD's. Haven't had any issues with refresh rates in PAL or NTSC. Low res works fine on my 3" LCD but high res does not because that screen is too low res. So I guess just have to live with swithing screens for output or get a medium res screen dedicated for use with this.

closed 2016-12-28 by Televicious