mist-devel/mist-board #42: NES core missing sounds in Super Mario Bros 3

vanfanel, 2016-12-03:
Hi,

Has anybody else noticed that the NES core is missing some SMB3 sound effects?
-The sound effect of Mario materializing on the map screen is missing.
-The sound effect of Mario flying with the tanooki or racoon suits has missing notes.

inDigiNeous, 2017-05-13:
Yes. Noticing some missing sounds on other games too, like in Lifeforce when you destroy the grapper like things in the first level.

vanfanel, 2017-05-24:
@inDigiNeous : Do you have any idea on what the problem could be?

greyrogue, 2018-04-03:
The issue is here:
https://github.com/mist-devel/mist-board/blob/ae8e19adf1285b091798a3667ae393877b627910/cores/nes/src/apu.v#L74
Instead of:
wire ValidFreq = Period[10:3] >= 8 && (SweepNegate || !NewSweepPeriod[11]);
It should be:
wire ValidFreq = (|Period[10:3]) && (SweepNegate || !NewSweepPeriod[11]);
The test is for Period >= 8, but the code is accidently, effectively testing for Period>=64.

sebdel, 2018-04-03:
@greyrogue, you are my hero.

vanfanel, 2018-04-03:
@greyrogue : Awesome! It seems to work perfectly well, Mario 3 sounds are complete now!!

Also, do you see a flickering area on the upper-left part of the screen?