mist-devel/mist-board #45: Amstrad Core Issues

HartOz, 2017-03-03:
The core does not support the the bundled CP/M+ software.

With a valid working CP/M+ Disc1 image mounted,  the systems returns with the following message after issuing the |cpm command.

"This program will not run in this environment. Press any key"

Also there seems to be no way of adding a second floppy disk drive to the core.

renaudhelias, 2017-03-04:
Nice, added to TODO-list (I've got exactly the same message in one-disk version of "Batman Forever" demo (two separate disk version runs fine))

Second disk drive is implemented, running nice on some (but perhaps have to insert disk in both drives before trying second one), for sample STarKos run fine using two disk, and SymbOS also.

FDC has problems (it throws "Bad Command" instead of "Not Found")

renaudhelias, 2017-03-23:
Shown as solved about CP/M+, the message "This program will not run in this environment. Press any key" is due to using bad language localization, see cpmplus.zip at http://www.cpcwiki.eu/index.php/System_Disk

But I'm working currently in some FDC improvements (r005.8.15)

closed 2017-06-04 by renaudhelias