mist-devel/mist-board #46: Tools / ESCParser

sblendorio, 2017-05-05:
I'm using ESCParser tool:
https://github.com/mist-devel/mist-board/tree/master/tools/ESCParser

It could be very useful, because it's compatible with "EpsonQ" driver from AmigaOS. But, I tried to convert a graphic output and I got this:

[image]

It seems like it have done more passes but interleaved.

[b.ps.zip](https://github.com/mist-devel/mist-board/files/978811/b.ps.zip)

sblendorio, 2017-05-05:
this is the original output from EpsonQ driver
[bb1.zip](https://github.com/mist-devel/mist-board/files/978825/bb1.zip)

sblendorio, 2017-05-05:
Moreover, the boundaries seem to be wrong: the upper part of the screen is cut out.

nzeemin, 2017-08-20:
@sblendorio, the ESCParser utility based on ESC/P specification for Robotron CM 6329.01M printer. Probably, the Amiga "EpsonQ" expect a bit different dialect of the ESC/P standard.

harbaum, 2017-08-21:
We saw the same kind of problems with the real thing back in the days.

I suggest trying to detect the amiga generated code and make some adjustments then. But please make sure that the atari st part still works.