mist-devel/mist-board #49: Configure mouse sensitivity in mist.ini or any other way

Floaf, 2017-06-13:
Is there a possibility to configure mouse sensitivity in any way? When i run minimig the sensitivity is to high so it's almost impossible to use.

VK3FNG, 2017-06-15:
Can I also add my support for this? It's driving me bonkers.

AmigaIstMiST, 2017-06-15:
I did a feature request for this in Rok's repo a year ago. However, it looks like he doesn't have much time for Amiga stuff recently.

harbaum, 2017-06-15:
This issue is mouse specific. A different mouse will probably give you a better experience.

This also isn't core specific. The mouse is likely too fast in e.g. the Atari ST core as well.

So the best way to address this is the arm controller firmware and not e.g. the amiga core.

I'll have a look at this.