mist-devel/mist-board #50: SPI speed

jotego, 2017-08-06:
This is a documentation request:

I am writting a core and using the SPI modules to communicate with the microcontroller, as expected. I do not know what frequency the microcontroller is using to send the data, so I cannot constraint the clock in my core for synthesis. The **frequency of SPI_SCK** is what I need.

Could someone tell me? And, could someone update the Wiki on this page so this information is available?

Apart from the SPI clock speed, do we need to set up any other constraint type to deal with latencies through the pins or the SPI clock edge is already on the stable side of the data waveform?

Thank you

renaudhelias, 2017-08-06:
MiST-board offers a SDCARD SPI emulation if your core is using SPI about SDCARD access :

Project using both "sd_card.v" and "user_io.v" controllers are in this way. The SDCARD SPI emulation does need our own spi clock (not SPI_SCK) that is enough low (as normal SDCARD CLK...) to be "strobe" throw theses controllers.

SPI_SCK is for MiST-board controllers : user_io.v, data_io.v, osd.v. It's a personnalised SPI : a bridge between firmware (ARM) and MiST-board controllers. So SPI_SCK is not really "end user" (it is more MiST-board specific)

jotego, 2017-08-06:
Thank you. I am not using the SD card at the moment. The firmware sends the file [core name].rom via SPI on start up so I use that. For that case, the SPI_SCK comes from the microcontroller but I do not know the frequency. 

I am getting the file transfer ok but I cannot read the joysticks so I wonder if I am doing something wrong with the SPI.

renaudhelias, 2017-08-06:
user_io.v 's statut[0] is "reset" saying when system is full ready before using controllers.

harbaum, 2017-08-06:
The spi clock is usually 24mhz. This depends on the sd card as it shares the same spi. Normally with a regular sd card its 24mhz. With e.g. an mmc card it will be lower.

I'll check the active data edge this evening and also post it here.

renaudhelias, 2017-08-06:
Last time I got problem it was about wiring controllers themself and with pins.
On my project :
Got user_io.v SPI_CLK in SPI_CLK board pin
Got user_io.v SPI_MOSI in SPI_DI board pin
Got user_io.v SPI_MISO in SPI_DO board pin
Got user_io.v SPI_SS_IO in CONF_DATA0 board pin
Got data_io.v sck in SPI_CLK board pin
Got data_io.v sdi in SPI_DI board pin
Got data_io.v SS in Vcc...

harbaum, 2017-08-06:
There are plenty of MIST tutorials demonstrating all the topics incl. one of the first ones explaining how to access the joysticks. None of these work for you?

harbaum, 2017-08-06:
Data is valid on the rising clock edge and is being set on the falling. You can also see this in the user_io.v

jotego, 2017-08-06:
@harbaum yes, I am looking at the tutorials. Indeed I have copied the data_io.v and user_io.v from the tutorials but I do not get the joystick inputs. Of course, I have checked the same joystick with a different core. I do get the ROM downloaded correctly. Note that the tutorials do not set any constraint on the SPI clock so that's why I was initially suspicious on that.

@renaudhelias I am going to double check pin connections. By the way, even if status[0] is not set/clear, the only thing I can do is wait, isn't it?

renaudhelias, 2017-08-06:
status[0] is reset ('1' does reset)
For me (in my project...), system is ready when status[0] has falling down (from 1 to 0) so then I can do things.

jotego, 2017-08-06:
So thanks to your clues, I have discovered how to use the status[x] signal. I have used it succesfully to modify signals inside the core. However, when I connect the joystick signal to where I connect the status signal, I do not get a reaction. Is there any reason why the firmware should not send the joystick commands but still send the status bits?

renaudhelias, 2017-08-06:
Perhaps you inversed your joystick ports ?

jotego, 2017-08-06:
I am using the service mode so I can see both inputs. But I will double check!
[image]

renaudhelias, 2017-08-06:
Not all games are using joystick...

renaudhelias, 2017-08-06:
The red led of MiST-board is on pin "LED" :)

harbaum, 2017-08-06:
You need to return a valid 'core type byte' via spi on request. Otherwise only very few services are available via spi. See user_io.v in the tutorials.

This was introduced because the firmware offers some core specific support for the amiga and the atari st. You should likely use '8 bit core type' like the tutorials also do.

jotego, 2017-08-06:
@renaudhelias in the end it was a misconnection due to renaming signals from joystick_0 to joystick1 at some level of the hierarchy. Thank you!

closed 2017-08-06 by jotego