mist-devel/mist-board #52: Noise while loading from SD-Card

uchristo, 2017-10-17:
I'm having strange Sound issues when the SD-Card is being accessed. Seems to be some kind of crosstalk?

harbaum, 2017-10-17:
Is that with a specific core? Some cores like the spectrum one do that by purpose since the original machine also made some noise when loading stuff e.g. from tape.

uchristo, 2017-10-17:
It's in the MiniMig AGA Core..... I doubt, that this is "normal" Operation.
Could it be a ground-loop? But in that case I would expect the problem to be permanent (not only while file IO)

uchristo, 2019-12-12:
Still got this Problem. I don't know if it's because of a ground-loop?

My Connection: MiSt from (3,5mm Jack + VGA) to SCART into OSSC, from there via HDMI to the Television. The Television outputs the audio via Optical SPDif to the AVReceiver.

MiST and OSSC are both powered by the same Powersupply. I tried to output the 3,5mm Output from MiST directly to the AV-Receiver but get the noise as well. The only modifications I made to the MiST-Board was to remove the Resistor for Stereo-Mixing and adding a DC Power-Jack

harbaum, 2019-12-12:
Do you have that issue with other cores as well? That would tell us if it's a hardware problem. In general are the SD card and the sound output technically pretty unrelated. They aren't even connected to the same chip and are on opposite sides of the case. I ground loop would not show any relationship to the SD card.

Do you also see e.g. video distortions at the same time?

uchristo, 2019-12-12:
I‘ll try wirh another core and do some testing with just audio Connected

uchristo, 2019-12-15:
Hadn‘t much time, so I tested with just Audio and Power connected. noise while Loading is still noticable. I‘ll plug to my Oscilloscope to identify the Frequency. I changed the   Powersource, too.

uchristo, 2020-01-06:
Update: I tried the same Power-Supply using Micro-USB. It's definitely a ground loop.
When attaching the DC Power-Jack, I drilled the hole in the enclosure about half an millimeter wider than the plug itself but did not exactely match the middle. The ground-plate from the Plug is connected to the ground from the PCB _and_ to the enclosure. Next I'll remove the board from the Enclosure and retest with the DC Plug.
Nevertheless, when loading Data from SD, there is still some frequency-variing noise present even with Micro-USB for Powering.

uchristo, 2020-02-18:
Is there a special reason for why the housing is not connected to ground? The PCB-holes are covered with paint on top and bottom....

harbaum, 2020-02-18:
You need to ask Lotharek about these things. The design of the case has been done by him.

uchristo, 2020-02-18:
But the PCB Layout itself doesn‘t connect Ground to the Drill holes

uchristo, 2020-02-18:
But the PCB Layout itself doesn‘t connect Ground to the Drill holes.... this is quite unusual....