mist-devel/mist-board #54: Only ST core works!

Galbi3000, 2018-07-10:
I just got my MiST 1.3 yesterday but so far I am only able to get it to work with the ST core. I have tried multiple versions of the Amiga core, the Spectrum core and the BBC core, they will not work.

If I boot with the ST core and try switching to the Amiga by selecting the Amiga core file on the SD card with a valid kickstart 3.1 ROM stored as kick.rom (I own Amiga Forever Plus and have licensed ROM files. No, I am not using the Amiga Forever ROM as it is encrypted with a key file, I have actual ROM files downloaded elsewhere). When I switch core using the OSD menu Firmware & Core sub menu the red LED flashes while it reads the core file then the amber LED comes on for 2 seconds then flashes 5 times before repeating that sequence of 2 seconds on the flash 5 times. The only information I can find on this site is that the red LED should be flashing for errors and 5 flashes means the core was not read properly.

I get the same result if I switch to the Spectrum core or the BBC core in the same way. Again the same result if I also switch the main ST core.rbf file with the Amiga one (rename them so the Amiga one is core.rbf and the ST one is another name). Again the same result. All cores I have tried (that I have system ROMS for) have the same result except for the ST core!

I did update the firmware to ATH170412.

I would like this to work. I spent a lot of money on getting this on the premise that it can be used to 'emulate' (yes I know it's not actual emulation) multiple machines. In particular ST AND Amiga. If I just wanted an ST I would have spent less getting a real ST :P

Galbi3000, 2018-07-10:
I think I get what is going on. My TV is able to view ST mode but I think it is having problems with refresh rates from other machine cores! I will have to invest in a multi-sync monitor :)

closed 2018-07-10 by Galbi3000

harbaum, 2018-07-11:
That maybe. Don't you have a regular VGA screen? Most cores work out of the box with these.

My first assumption was that you are downloading the cores the wrong way. If the 5 flashes also happen with the spectrum and other cores i'd strongly suggest to check that the downloaded core files are fine.

Newsdee, 2018-07-15:
It's a common gotcha with Github that one may accidentally download an html
page instead of the core binary. You need to click on "view raw" to get the
right file.

Galbi3000, 2018-07-16:
That is good advice Newsdee however I got the core files by cloning the project to my computer :)

Galbi3000, 2018-07-16:
I discovered by accident that the issue is the TV/Monitor. It displays the ST resolutions fine but other screen modes used by other cores are incompatible. I have a multisync monitor arriving later today that is guaranteed compatible with Amiga modes, the model was tested by Amiga users and listed online as a good choice. I hope that means it will also work with other cores too :)

Galbi3000, 2018-07-17:
The monitor works fine. Some cores do not. In particular the Amiga cores. Spectrum, QL and BBC work fine but I can not get Archimedes and Amiga to work. A shame as the Amiga is one of the main 2 machines the MiST is suposed to clone!

sleary78, 2018-07-17:
Do you have the SD card setup with the correct ROM? That’s the easy way to fail here

Galbi3000, 2018-07-17:
It seems it did not like the Kickstart 3.1 ROM, my 2.04 ROM works fine. OK. Now I think I have a grip on what it going on. I will try other copies of the 3.1 ROM to see if the one I was using is corrupt or if the core just does not like 3.1 :)