mist-devel/mist-board #55: Archimedes floppy files

Galbi3000, 2018-07-11:
In the instructions for setting up for Acorn Archimedes you state the following:

"Floppy disk images ADF format and of exactly 819200 bytes in size are currently required."

How do I make those? I have WinUAE and it will only create specific fixed sizes, none of which are 819200 bytes!
The closest is "Create Standard Disk" with "3.5" DD (PC)" selected which creates a disk file of 737280 bytes (720KB).
The next closest is "Create Standard Disk" with "3.5" DD (PC)" selected, creates a disk file of 901120 bytes (880KB).
After that all the other options are too big apart from one that is smaller than the 3.5" DD (PC) one.

sleary78, 2018-07-11:
It’s not an Amiga adf file. It’s acorn adf. Use ArcEm or Arculator

Galbi3000, 2018-07-18:
OK, I got an Archimedes blank floppy image from an Archimedes archive web site and copied that as floppy0.adf and floppy1.adf but the core fails to start up, just a black screen.

sleary78, 2018-07-18:
> I have a number of versions ranging from 3.0 to 3.6 but I used the 3.0 version for MiST as the Wiki says that version was tested and known to work with the core.

Nothing above 3.0 will work. 2.0 and 1.x will. 

3.6 is not for the Archimedes

Galbi3000, 2018-07-18:
The notes for the Archimedes core ( https://github.com/mist-devel/mist-board/wiki/CoreDocArchimedes ) state that only RISCOS 3 supports VGA, that suggests to me that version 3.0 at least has been tested. That's the version I used, not 3.6! Yes I said I have versions 3.0 through to 3.6 but I did say I only used 3.0

sleary78, 2018-07-18:
I know. I wrote the core ;) 

RISCOS doesn’t default to VGA. Try plugging headphones into the audio jack and see if it “beeps”. If so hold down 0, 4 or 5 on the numeric keypad during a cold boot. That’s how the OS figure out mode to boot in

Galbi3000, 2018-07-18:
I put it on it's own SD card with the following files:
core.rbf         319 KB   - Archimedes core
riscos.rom     2048 KB - RISC OS 3.0 ROM
floppy0,adf   800 KB   - Blank Archimedes disk image from Archimedes archive web site
floppy1.adf   800 KB   - Another blank disk image.
And a folder with a few Archimedes game disk images from that same archive site.

At boot the red LED flashes while it reads the core, I hear the audio pop as the machine resets the audio. No beep, no display. Nothing, not even the red or amber LEDs flashing for an error :/

I will try to get hold of a 2.0 ROM image and try that.

sleary78, 2018-07-18:
> Same result with RISC OS 2.0 ROM

Then there is some issue with your MiST. Some of the newer ones had
different RAM IIRC?

sleary78, 2018-07-18:
> One thing I have noticed is that ALL documentation assumes the MiST is the older model without MIDI ports. One difference in the documentation is that the docs say the red LED should flash error codes but mine it seems to be the amber one doing that function!

I’ve only ever seen the early prototypes built by Till by hand. Never seen a later model.

harbaum, 2018-07-18:
The MIDI part wouldn't make a difference with any core. It's just not used in most cores.

The RAM chip did change as the old one wasn't available anymore. However, both have identical specs which usually doesn't make a difference. But sdram timing is quite delicate and some cores are operating the ram a little outside spec violating setup and hold times by a few picoseconds or so. That's where the ram chip may make a different as it's unspecified how they react on this.

The acorn core and also the msx core are known to be a little bit tricky here. I have recently been told that increasing sdram drive power from 4 to 8 mA gains a few ps. That may actually help ....

hubersn, 2018-07-19:
I am a bit confused about

> Nothing above 3.0 will work.

Surely you mean 3.1x? I have successfully used RISC OS 3.10, 3.11 and 3.19 (the German variant) without problems.

Currently I wonder about how CMOS is implemented (or if it is implemented at all, it seems to lose all configure'd stuff after reboot). I was not successful in forcing the core to anything that would drive 1 15kHz screen, because a configured monitortype is thrown away, pressing the key on the numeric pad on boot does not work and I don't know where in the RISC OS ROM the CMOS defaults are stored (or if they are programmatically created somewhere).

hubersn, 2018-07-20:
OK, found the CMOS content in cmos.mif in the Archie sources, which explains where the default values are coming from.

terriblefire, 2019-02-15:
Read the RISCOS docs. You can hold down various keys at startup to reset CMOS settings for VGA etc.

hubersn, 2019-02-15:
> 
> 
> Read the RISCOS docs. You can hold down various keys at startup to reset CMOS settings for VGA etc.

Have you actually tried to do this? It never worked for me on MIST. And CMOS settings changed by configure commands are not saved back - there is no CMOS file, and the ROM is (thankfully!) not updated.

It could in theory be a timing issue with my USB keyboard of course that early keypresses are not correctly reported to the core. No idea about the inner workings of either the Archie Core or how MIST firmware deals with the USB side of things.

sleary78, 2019-02-15:
Yes. It works until you power down and back on again. 

Between power cycles use Ctrl break to reset and get the cmos changes

gyurco, 2020-06-16:
Lot of changes happened in the Archie core since this issue opened, I think it can be closed.

closed 2020-06-16 by gyurco