mist-devel/mist-board #88: Mice on atari core

gtranche, 2019-03-14:
Is there any news about mouse support on the atari core like on the minimig core ?
I tried som rs232 to usb cable with no success and i would like use my old mouse with my mist :)

gyurco, 2019-03-20:
What kind of mouse is it? The original Atari mouse AFAIK is not RS232, but using quadrature motion detection on the joystick direction pins. That wouldn't be that hard to implement with the DB9 joystick ports.
But is there any value of an RS232 PC-mouse? What would motivate a developer to support those old cheap ball mice?

harbaum, 2019-03-20:
The original Atari ST mouse delivers the same quadrature signal as an amiga mouse (only the wiring differs on two pins). This is evaluated by the so called ikbd processor in the Atari ST. The ikbd in turn is _not_ implemented in the core but in the MIST firmware (once i get really bored i'll implement it in the core). So this feature would have to be implemented entirely in the firmware. The reporting to the core would use the exact same logic that's currently using to send the signals derived from the USB mice.

A few things would have to be made configurable since the atari/mist cannot tell if there's a mouse connected or not.

harbaum, 2019-03-20:
And yes, these are ball mice and they were prone to major defects. I doubt man of them exist in a satisfying state.

gyurco, 2020-06-16:
Original Atari mice reported to work in the MiSTery core.

closed 2020-06-16 by gyurco