mist-devel/mist-board #105: [C64] question about pll reconfig

nippur72, 2019-04-29:
Not a real issue, perhaps a small simplification in the code.

I noticed the C64 core (and VIC20 and I suppose some other) has a pll reconfiguration mechanism for handling the case PAL/NTSC. I wonder why this has not been implemented simply as two outputs pll (`c0` and `c1`) since the Cyclone chip allows up to 4 clock outputs for the same pll.

Is there any particular reason?

And in case, is it worth to remove the pll reconfiguration and simplify the code? The selection of the frequency could be as simple as 
```
assign clk_sys = st_ntsc ? clk_sys_ntsc : clk_sys_pal;
```

harbaum, 2019-04-29:
That's how it previously was. The problem is that you then have a combinatorial outout as a clock source which imho limits the usability of the dedicated clock networks inside the fpga. Using the reconfigurable pll allow to riute it's outputs directly to a clock network.

nippur72, 2019-04-29:
ok, perfect thanks, that's a good reasonably reason. Still learning :-)

closed 2019-04-29 by nippur72

sorgelig, 2019-04-30:
Actually, with simple 2 clocks selection like this Quartus includes special module "clock selector". At least how it's happened on Cyclone V. Since the intrinsic function has "cyclonev" in the name i'm not sure if the same function exists on Cyclone III device or it's Cyclone V specific. So, basically it may work. But the key point is precise PAL and NTSC clocks. It's impossible to use a single PLL with both these clocks with precise values. So 2 PLLs are required in this case.

nippur72, 2019-04-30:
@sorgelig talking of precise clocks, is it possible to achieve exact frequencies (at 1 Hz resolution) by combining two plls in serial connection? E.g. the output of the first pll as input to a second one?

sorgelig, 2019-04-30:
Yes, you can.