mist-devel/mist-board #109: Minimig core: error #80000004 upon boot when using MMULibs

igully, 2019-05-08:
I have been witnessing consistent minimig core error #80000004 upon boot, when using MMULib (http://aminet.net/util/libs/MMULib.lha) together with AmigaOS 3.1.4.

I contacted the MMULib author and he told me:

..."This alert is an "illegal instruction", probably indicating that MiST
does not recognize a valid 68020 instruction. Under emulation, I do not
see any issue. Thus, it would be helpful to find which instruction
exactly the MiST does not like, and then find out where the problem is.

If I'm allowed to make a short into the dark: Check for movec, and the
source and destination function code registers. They are rarely used,
though important for proper usage of the MMU."...

sorgelig, 2019-05-08:
bare 68020 has no MMU.
read the requirements for this library.

igully, 2019-05-08:
The mmu.library does not require a MMU. It provides an interface for a MMU
only if available, and also attempts to workaround cpu specific microcode
bugs.

sorgelig, 2019-05-08:
in readme: http://de6.aminet.net/util/libs/MMULib.readme
Requires:     a 68020/68851, 68030, 68040 or 68060 with a working MMU
so you need 68851 (MMU) which is not implemented.

igully, 2019-05-08:
From the MMU.guide included in the MMULib archive:

Compatibility: The compatibility guideline -> 68020 based systems

"Systems with a 68020 CPU may or may not have a MMU. Since the 68020 does not
come with a built-in MMU, an external chip, the "68851 MMU" is required. If
this chip is not available and not installed on your system, it doesn't make
much sense to install the mmu.library. **It won't fail**, but it just wouldn't
be able to do much for you, see"...

The thing is that with the set of MMULib libraries installed, the 68020.library makes the Mist produce an Amiga error  #80000004 upon boot, when it should not produce anything.

If you test the MMULib set, in a real Amiga 1200 or a WinUAE emulated one (both without an MMU), it boots without issues. On the Mist, on the contrary, it errors.

sorgelig, 2019-05-09:
I see. 68020 model in Minimig is not perfect, yes.

retrofun, 2019-09-23:
Doesn't crash anymore with the implementation of RTD instruction to TG68K:
https://github.com/retrofun/minimig-mist/commit/c4694d5a1fddf12f808d226a01826fd486467af4

igully, 2019-09-24:
@retrofun 

Thank you for implementing it. 
So this bug should be closed then.

closed 2019-09-24 by igully