mist-devel/mist-board #110: SLIP on 1.3 board

sardaukar, 2019-05-21:
I found this https://docs.google.com/document/d/1FlnxLtpxHlDc56A07uFv0sb5GYEPq3SvTZ7YbWXeQyw/edit guide for ESP8266 wiring up inside a MiST and I'd like to try it. I don't have the version with the MIDI ports, so what are the pins I should connect to the ESP as RX and TX on the 4-pin serial header on the left of the v1.3 board I have?

Thanks!

gyurco, 2019-05-22:
https://github.com/mist-devel/mist-board/wiki/Board13

1-GND, 2-3.3V, 3-RX, 4-TX

harbaum, 2019-05-22:
Exactly. And when you sit in front of your box this counts from front to rear. The pin closest to the front is GND.

harbaum, 2019-05-22:
Also the directions are as seen from the FPGA. So TX is used by the FPGA to send data to the ESP and RX is used to send data from the ESP to the FPGA.

sardaukar, 2019-05-22:
Can I power the ESP8266 from the serial port too? The guide suggests plugging into the voltage regulator.

gyurco, 2019-05-22:
Just checked the ESP8266EX datasheet: the radio part's max. current can be 150 mA, probably it's too much for the 1A LM3940.

sardaukar, 2019-05-22:
So I have to use the voltage regulator like the guide suggests?

gyurco, 2019-05-22:
If you overload the internal regulator, it's protection will lower the 3.3V output - you'll get instabilities. If the ESP will really overload it - I don't know, but it's possible when using the radio.

sardaukar, 2019-05-22:
Ok, so the LM3940 port is 1A, but is the 3.3v on the serial header 1A too? If so, how easy is it to lower it to 150 mA?

gyurco, 2019-05-22:
You're mixing current with voltage. Current depends on the impedance of the device. 1A is the maximum current which the LM3940 can supply, you must not exceed it with the FPGA internal + external consumption.

sardaukar, 2019-05-22:
Ok so will I blow up the MiST if I hook this up to the serial headers? :)

gyurco, 2019-05-22:
Don't think so, maybe it'll be just unstable during heavy radio-using.

sardaukar, 2019-05-22:
I see. Maybe that's why they say 38400b/s is the maximum stable speed, even though it can do 115200b/s. Thanks for all the help, I'll try it and report back :D

closed 2019-05-22 by sardaukar

sardaukar, 2019-06-14:
I managed to get it online and have blogged about it https://blog.nootch.net/post/amiga-bbs-online-2019/ :D

harbaum, 2019-06-15:
Wow! I really enjoyed this blog posting.

This is the kind of spirit that made me build the MIST in the first place: Tinker with modern hardware to revive the old setups.

And you are powering the esp from the MIST? That should be fine. You'd notice instabilities when you overload the power regulator. If everything works stable then you are just fine. If the system runs for a while and you touch the voltage regulator in the MIST and It's at most slightly warm then you are super fine as the power supply is far from being overloaded.

And a final remark: The MIST still seems to be produced as Lotharek and some of the vendors recently show it as being available again.

sardaukar, 2019-06-15:
Oh, good to know - I'll update the blog post. I saw this http://www.atari-forum.com/viewtopic.php?t=32998 and thought it was done. Also, the Amiga AGA core hasn't had a significant release in a while so I thought everyone had moved on to the MiSTer.

sardaukar, 2019-06-15:
@harbaum I just had a look, it seems like the 1.4 version being sold at Lotharek no longer features the serial port header :( https://lotharek.pl/productdetail.php?id=97 Oh, and by the way, thanks for creating this project - it's really awesome!

harbaum, 2019-06-16:
Indeed, the 1.4 does not have that connector. But the signals are still available somewhere. I'll write a wiki page about that and report here.

gyurco, 2019-06-16:
But the MIDI IN is just decoupled with an opto-coupler, and OUT goes through 2 inverters. Probably they can be used directly (somebody reported the tape input works with the MIDI IN connector).

sardaukar, 2019-06-16:
Would be good to get a new guide to achieve the same SLIP connection with the 1.4 board!

harbaum, 2019-06-16:
You likely go through the MIDI with some additional parts as 38400bit/s is only slightly faster than the 32500 bit/s of midi.

Bit by cutting a trace or two you can connect the esp internally without any further parts.

gyurco, 2019-06-17:
MIDI OUT pin 5 just the FPGA TX goes through 2 Schmidt-triggers, I don't think it limits the speed. Which is not very fast, as you stated.
MIDI IN pins 4-5 just drives the optocoupler, I also don't think it'll limit the speed of the serial port (MIDI is a serial port at the end, it's even implemented via UART chips - the ACIAs).
I would try it before damaging the board :)