mist-devel/mist-board #111: Replace EP3C25E144I7N with 10CL025YU256I7G

moozoo64, 2019-06-13:
I just wanted to suggest that a cheaper more modern fpga be used.
On Mouser the EP3C25E144I7N  costs $108 AUD and the 10CL025YU256I7G costs $38.48 AUD
Mouser doesn't list the max operating frequency so maybe that's an issue?

harbaum, 2019-06-13:
Are you planning to build an updated MIST yourself? Or are you suggesting someone else doing this?

The newer chip sure is nice. But a new MIST based on that would require major updates to all cores and the new board would be incompatible with the old one.

gyurco, 2019-06-13:
Seems these chips have the same specs in LCs and BRAM, if you want to replace the Cyclone III, then it would be wise to put there a larger FPGA, the cores have to be recompiled anyway. I wouldn't build a new MiST without at least 2 MBits of BRAM and 55K LCs.

sorgelig, 2019-06-13:
MiST uses EP3C25E144C8N which is almost twice cheaper than EP3C25E144I7N.
And yeah, FPGA is not CPU, you can't replace it without modify re-compile all cores.

moozoo64, 2019-06-13:
@sorgelig 
>MiST uses EP3C25E144C8N which is almost twice cheaper than EP3C25E144I7N.

Mouser has the EP3C25E144C8N  at $72.38 and the 10CL025YE144I7G (144 pin version) at $40.02 AUD So it still much cheaper. And I'd imagine most of those 144 pins would be gpio so its mostly pin compatible (?)

>And yeah, FPGA is not CPU, you can't replace it without modify and re-compile all cores.

So there is a lot of difference between these fpga's on the inside? There have the same specs.
Yes I'd expect a recompile.

@harbaum 
>Are you planning to build an updated MIST yourself?

I wish, maybe when I retire I'd have the time.
I want to know how practical it would be to make a cheapish low powered mist on a m.2. slot card. Two fpga's. The other one handling the pcie and capturing the vga out to the hosts memory.  Something like a PicoEVB + Mist
The 8 of these go into a double sided 8 slot pcie16 m.2. adapter,
And 8 of those go into an AMD Epyc server (128 pcie lanes)
And 1000 of those servers go into a data center.
Think Google Stadia for mist.
But... you can can get more than just sdd's on m2. i.e. crypto currency mining cards, capture cards, Also think limesdr, dslams and so on.
Lenovo  put an Aria 10 GX on an M.2. slot card https://3s81si1s5ygj3mzby34dq6qf-wpengine.netdna-ssl.com/wp-content/uploads/2018/04/image003.png
Anyway I'm now totally off topic.

You can close this issue. Or mark it and an enhancement request for the existing mist hardware.

harbaum, 2019-06-14:
We know that there are alternatives. But the big question is: What would be the benefit? A bigger FPGA would sure allow for some new cores that are not possible with the current MIST. The MISTer has some of them. But is that really worth de-doing all cores and e.g. risking that support for the old board will stop? I personally don't think so.
And developing a new unit into a state where you can actually sell it is not cheap nor simple ...

closed 2019-06-14 by harbaum