mist-devel/mist-board #114: Feature request: multiple file load

jotego, 2019-08-20:
Arcade cores need a cumbersome step to the user: convert original ROM files to a single file. We are all familiar with that.

Recently, Bruno Silva et al. have developed a system based in XML files. Each file contains the list of files to be sent to the FPGA and the order. The user just needs to unzip the original ROM set from MAME into the SD. The file browser shows this .xml files as items to load them and then the microcontroller reads them and sends the appropriate content.

harbaum, 2019-08-20:
The MIST has somerhing similar for quite some time. You can specify rom entries in the ini file and upload multiple files subsequently. I have introdruced that for the c16 core which also uploads kernal, charset, basic, .... roms

jotego, 2019-08-20:
Thank you. I just read about it [here](https://github.com/mist-devel/mist-board/wiki/DocIni). It has two problems though:

1-MAME ROM files almost never end in ".rom". They don't follow any naming convention at all.
2-A single core may support several ROM sets, like the Japanese, the American, the booleg 1, the bootleg 2, etc.

I suppose point 1 is an easy change. For point 2, it needs a mechanism to let the core choose from the available entries in the INI file (and letting the ini file have more than one entry per core).

gyurco, 2019-08-21:
Maybe an 'L' option string which can be used in place of 'F', which tells menu.c to choose from .ini file defined romsets instead of a file mask would do the trick.
I'm sure an XML parser won't be added to the firmware.

jotego, 2019-08-21:
XML is an overkill for a microcontroller. I agree with that. If a ini file
with the name of the core can be browsed with an L option that will do the
trick. :-)

jotego, 2020-04-17:
I close this as we have now the ARC/mra system in place.

closed 2020-04-17 by jotego