mist-devel/mist-board #120: New MSX2 Game not working?

micv2, 2019-10-08:
Hi,
I bought a new MSX 2 game, but it does not run on the MSX core. It's called The Sorrow of Gadhlan 'Thur and is from Kai-Magazine Software.

https://kai-magazine-software.fwscart.com/The_Sorrow_of_Gadhlan_Thur_DIGITAL_VERSION_2019/p4398489_20023069.aspx

The game is a 2MB ROM. Is it possible to play the game?

Best regards,
MiC

gyurco, 2019-10-08:
Don't know. Is it working on the original One-Chip-MSX? Maybe you can ask this on msx.org.

micv2, 2019-10-10:
Hello, thanks for the answer.
I asked the question in the MSX forum …

https://www.msx.org/news/development/en/rpg-the-sorrow-of-gadhlan-thur-from-kai-magazine#comment-366914

harbaum, 2019-10-11:
This "One Chip MSX" is not MIST specific (https://www.msx.org/wiki/One_Chip_MSX). It was ported over to the MIST. So the question is whether the game works on those original OneChipMSX machines. If it works there it's a problem specific to the MIST port. Otherwise it's a general problem affecting all OneChipMSX based machines.

micv2, 2019-10-12:
Kai Magazine wrote: I know it works on one chip and Zemmix. So probably the ocmsx-core is outdated for TC64 and MiST?

gyurco, 2019-10-12:
This one:
https://github.com/mist-devel/mist-binaries/blob/master/cores/msx/msx_mist_190708.rbf
is based on the latest from KdL (3.7.1):
https://gnogni.altervista.org/
What's the error?

micv2, 2019-10-12:
I've overlooked that there are newer versions .. shame on me .. Have used Starter Pack V5. Have copied the linked core on the sd card, now the msx-bios is not loaded .. simply exchange the core file was not successful. I have to learn more about it, excuse me ;)

gyurco, 2019-10-12:
Here you can have a link for a msx.vhd file which has the correct BIOS:
http://www.atari-forum.com/viewtopic.php?f=115&t=37099

micv2, 2019-10-12:
Thanks, I will try later..

micv2, 2019-10-12:
What am I doing wrong, MSX BIOS does not want to be loaded .. Image stays black, core (F12) is loaded.
MSX BIOS has been copied on the sd first..

micv2, 2019-10-12:
Success, the game works on the MiST. Everything you need is here on the page. Setting Slot2 with MegaSCC + & 2MB. I'm excited, thank you :)
Have the .vhd- image directly copied to the SD .. Thanks :)

(Whether it works with TC64 too?)

sorgelig, 2019-10-13:
so, at the end it's just a matter of proper expansion settings in OSD?

micv2, 2019-10-13:
hello, not only. I still have an old core. Only after I got here the new core with the correct msx BIOS, the game run. Regards

closed 2019-10-14 by gyurco