mist-devel/mist-board #121: Apple II keyboard in Karateka

DFL1973, 2019-10-10:
I believe there is some kind of problem with the keyboard on the Apple II core. Karateka is impossible to play with the keyboard and even Joystick is working a little slow (but it's working). If I press "k" for keyboard, the fighter only executes a punch or kick after 5 seconds or so. Pressing "J" the joystick takes places and it´s possible to play again.

I already tried several disk images and the same NIBs on Applewin emulator works great with the keyboard.

gyurco, 2019-10-11:
Apple II+ or II/e core? They have very different keyboards.

DFL1973, 2019-10-11:
oh, sorry... IIe

gyurco, 2019-10-11:
Please try this:
https://github.com/mist-devel/mist-binaries/blob/master/cores/appleIIe/apple2e_191012.rbf

DFL1973, 2019-10-12:
Solved!!! 
Thanks!

closed 2019-10-12 by DFL1973