mist-devel/mist-board #124: 68881/2

UgoCaneFifone, 2019-10-25:
Hi,
can be supported the FPU 68881 or 68882 in the mist core ?

harbaum, 2019-10-25:
Sure. But someone needs to write a 68881 FPU. Can you do that?

Most people don't see much benefit in this since only very few software ever used that.

jotego, 2019-10-25:
That is a part it has always called my attention. But I am afraid to do it only to find out that no software uses it.

UgoCaneFifone, 2019-10-31:
Sculpt 3D/4D uses 68881
And the best would be to emulate an A3000:
The Amiga 3000 was designed as the third generation high
performance Amiga system by Commodore around 1988-89.  Unveiled in 1990
with much fanfare, (even a BYTE Magazine cover article) the A3000
incorporates a high performance Motorola 68030 CPU at 16 or 25 MHz, and
MC68881/16 or 882/25 floating point coprocessor into a full 32 bit system
architecture with 32 bit memory access.  In comparison to Intel systems,
the A3000's 68030 is about on par with a 80386.

sorgelig, 2019-10-31:
No, better is 68060@100MHz. Why you limit yourself? ;)

mattuna15, 2026-02-09:
I’ve started an open-source MC68881 FPU implementation in VHDL.

The goal is functional and timing-accurate behaviour suitable for FPGA-based 68k systems.

The project is underway and making solid progress. Documentation and test coverage are evolving alongside the core.

Repo: [Repo](https://github.com/mattuna15/68881-fpga)

Feedback from 68k / FPGA folks very welcome.

B) Functional Completeness (Core Missing Ops)
---------------------------------------------
[ ] B1. Implement FSQRT (square root).
    - Target cycle count ~120 (per datasheet guidance).

[x] B2. Implement FMOVE/FMOVEM family per programming guide section 3.3.1.
    - Register transfer (FPR<->FPR).
    - Move to/from memory for single/double/extended precisions.
    - Move to/from control registers (FPCR/FPSR/FPIAR).
    - Done: FMOVE for integer B/W/L source formats.
    - Done: FMOVE packed-decimal (.P) with static/dynamic k-factor.
    - Done: FMOVECR constant ROM fetch (#ccc,FPn).
    - Done: FMOVEM Dn bitmask mode and -(An) bit-order rules.

[x] B3. Complete dyadic operations set (guide section 3.3.2).
    - Done: FADD, FSUB, FMUL, FDIV.
    - Done: FCMP, FMOD, FREM, FSCALE, FSGLDIV, FSGLMUL.