mist-devel/mist-board #129: Load/save issue?

dfilskov, 2019-12-02:
This is an issue on the MiSTer core but I think it's an issue with this core too.

I've composed a bunch of tunes for Gameboy almost 20 years ago - but it doens't work when I load them into SRAM for the Carillon Editor (copying it to the sdcard overwriting an already existing save - carillon.sav ). The data doesn't seem to get loaded.

I've attached the .gb for Carillon Editor - and .bin and .sav files for one of my tunes for testing.

[gameboy-save-load-issue.zip](https://github.com/mist-devel/mist-board/files/3911110/gameboy-save-load-issue.zip)

The Carillon Editor author Alexi Eeben wrote about this issue:
"maybe it’s ”emulating” a wrong memory controller? The old flash carts were (if I recall correctly) of type MBC5"

In Carillon Editor ...
Load tune = click Start button on LOAD FROM SRAM
Play = click Start twice on BLOCK ORDER LIST
Save tune = click Start on STORE IN SRAM

Any ideas how to make it work?

gyurco, 2019-12-02:
Currently no RAM implemented in any cart space.

dfilskov, 2019-12-02:
Great :) - thanks for looking into this!

dfilskov, 2019-12-02:
Wait ... - it works!! - the SRAM file wasn't loaded unless I booted the ROM file again. My mistake!

Check this out :)
https://www.dropbox.com/s/k668zoxfbkdp03i/Evil%20Mirror%20winner.mp4?dl=0

closed 2019-12-02 by dfilskov