mist-devel/mist-board #131: What does the joy_key_map .ini option do?

manast, 2019-12-27:
I see this option is used in many example mist.ini files but I could not find any piece of documentation/explanation for it, could someone that knows please explain how it is used?

jotego, 2019-12-27:
It is in the [wiki](https://github.com/mist-devel/mist-board/wiki/USBJoystickMapping).

manast, 2019-12-27:
no it is not :). I did a search in the whole repo, it is not explained anywhere...

harbaum, 2019-12-27:
Google for 'joy_key_map mist' gives me this:

http://www.atari-forum.com/viewtopic.php?t=28174&start=25

You basically use this feature to map certain joystick button events to keyboard key codes. This allows to use joysticks on machines that only came with keyboards and where games were controlled using the keyboard.

manast, 2019-12-27:
Yes, I did in fact google it (of course!), and I read that thread, but I could still not figure out how it is supposed to work, notice that the thread mentions joy_key_map but never defines what it does.
What I want is maybe the opposite of this feature, i.e. I am using a IPAC-4 controller (which is basically a keyboard controller) but some of the cores only seem to work with joysticks (like Genesis or Snes).
So I wonder if there is anyway to create a virtual joystick by mapping keyboard codes to the virtual joystick pad and buttons, is it possible?

harbaum, 2019-12-27:
There already is a keyboard to joystick layer. Just press numlock and the cursor keys, shift and a few other keys are mapped to the joystick. I am pretty sure this is documented somewhere.

gyurco, 2020-01-02:
Interesting that almost nobody knows about the built-in joystick emulation. Seems nobody reads the scrolling message in the OSD.

manast, 2020-01-02:
as a side note, in many cases needing to press a key (numlock) to activate the emulation is not very convenient (arcade cabinet for example), it would be great to have it as an .ini option too.

manast, 2020-01-02:
> Interesting that almost nobody knows about the built-in joystick emulation. Seems nobody reads the scrolling message in the OSD.

Well, the most logical to me is to have the documentation in one specific place, like the wiki. If it is also in the OSD is fine, but it should be written in the wiki too. I would do it, but I do not know all the details of this functionality.

gyurco, 2020-01-02:
Putting it into to .ini would have a advantageous effect: you lose all your normal cursor and modifier keys. Also what's the use case for joystick emulation in an arcade cabinet? You should use normal joystick connection there I think.

manast, 2020-01-02:
no, I am using an IPAC-4 sa controller: https://www.ultimarc.com/control-interfaces/i-pacs/i-pac4-board/
By default it uses the MAME standard key mapping. Jotegos cores also do that. All problems will of course go away if all arcade and console cores used MAME standard mapping.

gyurco, 2020-01-02:
It's possible to modify the cores, of course. But help is needed, there are a lot of arcade cores out there (luckily almost all in one place). This would be an easy, but very boring task.

gyurco, 2020-01-05:
In these cores, I've added MAME keys:
https://github.com/Gehstock/Mist_FPGA_Cores/tree/master/Arcade_MiST/Midway%20MCR%203%20Monoboard
If it works well, I might do a generic arcade-inputs module, and slowly (very slowly) migrate the existing cores to it.

harbaum, 2020-01-05:
Those interested in this feature might consider donating a matching joystick to those implementing it. Just as an idea ...

gyurco, 2020-01-05:
Good idea ;)

manast, 2020-01-05:
well, the thing with mame is that a standard keyboard will do, no joystick needed :D

harbaum, 2020-01-05:
The problem with your last reply is that you sticked a price tag onto gyurco's time. Actually a pretty low one ....

He did already spend quite some time to fix the problems you have and he offered to do more. Sure he can test with a regular keyboard. That's what he likely already did. But the ability to test is not the main concern here. It's motivation

gyurco, 2020-01-05:
Fixing the controls for mame is also doable with a standard keyboard + freely downloadable Quartus. Actually did the common module, and changing a core to use it takes about 5 minutes after you know what to do.

manast, 2020-01-05:
well, I guess motivation comes from users using your stuff, and a growing and healthier community. I contribute as I can, as patreon, or reporting bugs. I am myself maintainer of several open source projects (quite popular If I may add), and most of the time I am implementing issues and fixing bugs that I never use myself, but features I figure many other users will use and that is enough motivation for me.

gyurco, 2020-01-06:
I don't know what size of the MiST users community is. But I know the developer community is very small. Probably can be counted in one hand. Simply no way to satisfy all wishes without help.

manast, 2020-01-06:
Maybe more tutorials and HOWTO's would increase the amount of developers. I can only assume but I guess that developing for the MiST is a great practice for somebody that wants to work with hardware professionally.

harbaum, 2020-01-06:
What tutorial or how-to do you think is missing? What how-to or tutorial would make you try to fix things yourself?

manast, 2020-01-06:
Something like a tutorial on how to create a complete simple arcade core from scratch. Like all the steps from getting schematics, to implementing the core, testing, synthesising, debugging, etc. Kind of, documenting your work when starting on a new core in such a way that somebody else could use it as a template when creating their own cores.

gyurco, 2020-01-06:
What's the problem with Till's tutorials?
https://github.com/mist-devel/mist-board/tree/master/tutorials
For a total beginner, a complete arcade core from scratch is probably too big step. Start with small things, like adding some keyboard mappings, little improvements to existing cores, then maybe do a bigger independent job. At least that's how I started.

harbaum, 2020-01-06:
But this all reminds me of something else: I need to fix the firmware to allow the usage of hid enabled arduinos which in turn will allow to use a cheap arduino micro pro or similar to implement things that behave like keyboards and joysticks. The IPAC-4 thing can then be replaced by an arduino.

harbaum, 2020-01-06:
Those tutorials include pong which in turn is the simplemost arcade you can think of. The soc example is a more complex machine and is much like an arcade machine. This should actually be all you need to get started. Please tell me if you find something missing.

gyurco, 2020-01-07:
USB HID shouldn't be a standard thing? Or Arduinos have some own dialect?

harbaum, 2020-01-07:
It's a standard but a pretty complex one. Unlike any other keyboard the arduino reports to be HID device of type "none" instead of "keyboard". But within the HID descriptor it then describes itself as a regular keyboard with the typical 8 byte report. This is now supported. 

A similar fix is probably needed for arduino based joysticks and mice. But the most common application are likely keyboard like devices especially for this mame controller like setups. I may actually add a description of an example setup to the wiki.

gyurco, 2020-06-16:
I think almost all cores in Gehstock's repo are moved to the common arcade-inputs module. If any core missing that, please open an issue in that repo.

closed 2020-06-16 by gyurco