mist-devel/mist-board #132: Schematic and the layout of MIST 1.3+ MIDI FPGA missing

0ric1, 2019-12-28:
I did not find any board layout or schematic of MIST 1.3+ / 1.4 MIDI FPGA in the repro, doesn't this exist?
My MIST case looks like this one https://amigastore.eu/2355-medium_default/mist-14-midi-fpga-computer.jpg and board like this https://lotharek.pl/upimages/mist14.jpg but the board is labeled with MIST 1.3+.
I am looking for the pins TX_FPGA and RX_FPGA.
Are there differences between 1.3+ and 1.4 MIDI?

gyurco, 2020-01-01:
It's just a normal MiST + [MIDI Addon](https://github.com/mist-devel/mist-board/wiki/MidiAddon).
MIDI IN pin 4-5 can be used as RX, MIDI OUT pin 5+GND as TX.

0ric1, 2020-01-01:
Thanks for the info. Is there another way to connect directly to TX_FPGA / RX_FPGA without the indirection over the MIDI port? I want to input / output line-in/out from a cassette recorder in order to load old games from cassette and able to save from FPGA on cassette.

gyurco, 2020-01-01:
I think the MIDI IN is a good casette adapter with that optocoupler. There was one guy on the forums who successfully used it.
Saving using the TX pin is not supported in any cores. Casette audio might goes out on the normal speaker output.

0ric1, 2020-01-01:
Thanks again, I search the forums for the adapter and give it a try.

gyurco, 2020-01-02:
The "adapter" is just to connect the tape audio to MIDI IN pins 4-5.

0ric1, 2020-01-02:
But line out is only 2 Vpp and between 4 and 5 is the diode (voltage drop of ~0.7V if it's a silicon diode) of the optocoupler that needs >5mA and a 220R (5V) or 33R (3.3V) resistor, so I think a interfacing circuit is needed? I think an amp e.g. LM324 is needed to drive the led of the optocoupler or am I wrong?
Is the MIDI on the MIST 3.3V or 5V?

gyurco, 2020-01-02:
The 6N138 datasheet states input current of 0.5mA, not 5mA.
The schematic for the tape interface to direct RX connection has only a single NPN transistor, and from experience I know it needs rather high voltage to operate (maximum volume on a telephone ear socket). I don't think the MIDI IN is worse. You can try it with a direct connection, if it doesn't work, you can still build a small amplifier. 
The MIDI IN is designed for 5V, since this is what the original ST uses, but should work in lower levels. MIDI OUT is 3.3V, since that's the FPGA's IO level.

0ric1, 2020-01-02:
I know the tape in (connected to cassette line out in the 83's) from the Oric 1, the signal goes through a condensator and resistor to a LM358 and then to a resistor to the base of a transistor that (C) is connected to the 6522A.

0ric1, 2020-01-02:
I found the schematics of the Oric-1 tape in interface, I'll rebuild it and use this for driving the LED in the optocoupler of the MIST.