mist-devel/mist-board #134: Approach to aspect ratio on VGA output

jotego, 2020-01-20:
I just wanted to hear your thoughts about aspect ratio. My approach until now for scan doubling in MiST has been rather straight forward: double pixels with a little bit of softening. Now that I am preparing the CPS1 core, this apprach is not going to work out (check out this [video](https://www.youtube.com/watch?v=LHfPA4n0TRo&t=2s)).

I was wondering if there is already some nice scan doubler in some MiST core which preserves aspect ratio. Or if good aspect ratio on VGA output is not an aspect well controlled in MiST so far.

harbaum, 2020-01-20:
I don't see the problem. The MIST uses an analog output just like the machines it replicates. You can use any pixel clock in combination with any number of lines per screen. So pixels will have whatever aspect ratio you like. If you use a digital screen (a screen with a discrete number of pixels horizontally) then it's up to the screen to sample the analogue signal at whatever rate it thinks is appropriate and map it onto it's own row of separate dots.

You can even connect an old analogue VGA screen to your MIST and there won't be any horizontal pixelization except what the core does. And this will hopefully just do exactly what the original machine did which in turn doesn't necessarily result in square pixels. The scandoublers we currently use simply repeat each line twice at exactly twice the original speed. Since the screen will cram whatever number of lines you throw at it into the same vertical space without affecting the image width you'll still end of with the exact same aspect ratio. This however depends on the behavior of the screen.

jotego, 2020-01-21:
Yes, for CRT output there is no problem.

It's the scan doubler that makes me doubt but I think you're right: the screen will adapt to the right width because of the line speed.

Thanks.

closed 2020-01-21 by jotego