mist-devel/mist-board #135: is the HT1080Z working?

rampa069, 2020-01-27:
Hi. i think he HT1080Z is not working. i have the rom in the SD root folder but after starting it  i only get "@" in the screen. (Also resintetized but same result)

also tried the rbf from http://joco.homeserver.hu/fpga/mist_ht1080z_en.html with the same result :-(

gyurco, 2020-01-27:
I have no idea, it works for me.

rampa069, 2020-01-27:
and the rom place is in the root folder, isnt it?

gyurco, 2020-01-27:
Yes, if you don't have any rom= lines in your mist.ini. Fsck your SD Card.

harbaum, 2020-01-27:
That would also be my recommendation. Use a freshly formatted sd card and put only the core (renamed to core.rbf) and the ROM into the cards root directory. Nothing else.

rampa069, 2020-01-29:
tried all. The problem i think are the new sdram timings. something similar happened to SNES core. Is working with older SDRAM chips.

gyurco, 2020-01-29:
The SDRAM is clocked only @42Mhz, and the controller is not particularly different than in any other 8 bit core. Are Colecovision/C16/C64/ZX81...etc are working?

gyurco, 2020-01-29:
Ok, maybe this one:
change the "set_locaton_assignment PLL_1 ...." in the .qsf to:
````
set_location_assignment PLL_1 -to clkmgr|altpll_component|auto_generated|pll1