mist-devel/mist-board #136: SCART video cable; 50Hz only?

boomlinde, 2020-02-04:
On TV output via SCART, the wiki notes:

> The video signal must be a 15kHz/50Hz PAL signal.

I have TVs that accept 60 Hz RGB over SCART. Is there something about the cable or the MiST itself which prevents using 60 Hz over SCART, is it an error in the documentation or is it simply rare that a TV can handle such a signal?

harbaum, 2020-02-04:
The cable is a pure passive thing and would simply forward any signal. However SCART imho only existed in PAL world and officially only supported PAL frequencies. So it depends on the TV if it also supports NTSC over SCART.

gyurco, 2020-02-05:
My TV doesn't have any problems with NTSC signal on SCART.

boomlinde, 2021-08-31:
I have been able to confirm that this works with TVs that support it.

closed 2021-08-31 by boomlinde