mist-devel/mist-board #139: MiST fails to powerup if other voltage sources are connected

jotego, 2020-03-11:
This is a problem I started seeing with JTAG but today I've seen it with USB too. This is what happens:

-If MiST is connected to a device that provides a voltage while it is powered off, then it is likely that it will not power up at the turn of power button.
-For instance: if the JTAG cable is connected to a computer that provides voltage, then a dim light will still be visible in MiST red LED and it may not power up
-If the usb signal comes from an USB device that may have power (like an USB splitter which may be taking its voltage from the other USB host), then MiST may fail to power up.

This is a hardware limitation and it would require hardware modifications to work. Probably some diode. I know we may not be able to fix this at this time of the product life but I wanted to have it reported.

Feel free to close it once relevant people have read it.

gyurco, 2020-03-11:
For me, the USB blaster don't cause this issue. Didn't try with a self-powered USB device.

harbaum, 2020-03-11:
You are not supposed to feed power into the MIST in any other way than the micro usb port. These JTAG ports are supposed to provide power to a adapter if it needs so. 

What you expect to happen if you power the board backwards? Yes, a diode here and there will prevent such things from happening. But things like the USB splitter you mention are simply in violation of the USB spec and should not do that at all. They'll likely not do any harm since they are too weak to power the MIST properly as you've experienced. But I would avoid these situations in general.

jotego, 2020-03-11:
Well, speaking now as my main profession of analogue circuit designer, protections are there to prevent problems when things that are not supposed to happen, happen. 

Anyway, this was mainly informative as I understand that there will not be board revisions.

harbaum, 2020-03-11:
That's true. But imho you need to differentiate. I don't think that internal debug connectors like JTAG need any real protection. People using that should somewhat know what they are doing and it might even be helpful to be able to power the board that way ... 

I certainly agree on the USB ports. The MIST never had a full blown USB power supply implemented with all the current limiting and ESD protection and whatever should be there on a real PC or the like. But that's one of the tradeoffs that allowed a single person to build and manufacture this thing. Still i don't expect any harm from that. 

There are other compromises like that. E.g. the MIST should have had a digital video output a proper USB host instead of the SPI attached MAX chip. It should have had real video DACs for audio and video also with all kinds of protections. It should have used DDR instead of SDR RAM etc etc ... There would be no MIST at all had I tried to implement all that. It's not perfect. But it's the best compromise I could implement under the given constraints.

And no, there will not be a redesign because of that.

jotego, 2020-03-11:
It's a great piece of hardware anyway. Let me just add a note in the wiki as troubleshooting for people who may experience this issue.

gyurco, 2020-03-11:
Btw, a MiST 2 with a bigger FPGA, maybe a stronger ARM (there are some more powerful MCUs with the same pinout - and even cheaper - as the current one), digital output possibility (only pins, no HDMI chip or other fancy things - those can be on an external board), maybe 512MBit SDRAM (even 2 pieces).
I know this is not a commercially viable thing currently, just dreaming.

closed 2020-04-02 by jotego