mist-devel/mist-board #142: Problem ZX80/81 Core

Julitium, 2020-03-21:
Hi.
The ZX8X core not works for me..... always black screen with no f12 or any other function, its looks freeze.
Test:
RGB Cable and screendoubler disabled.

Always fail.....All versions.

Thanks.

gyurco, 2020-03-21:
The ROM file is in place?

Julitium, 2020-03-21:
The ROM is in place! (zx8x.rom  12kb)
Not work :-(

P. D: zx48 core and zx48.rom (16kb) Works but its old.

harbaum, 2020-03-21:
Did you try a different screen? Maybe your screen simply doesn't cope with the video timing of a  zx81

gyurco, 2020-03-21:
Maybe would be better to have ZX81 as the default, not ZX80. At least the hsync is hardware generated on that.

Julitium, 2020-03-21:
Tim is in right..... At VGA monitor its Works!!!.
Now, i Will do the default in Zx81 Gyurco.
Thanks.

Julitium, 2020-03-21:
Gyurco is in Right.... Default to Zx81 Works fine!!!.
Thank you.

Julitium, 2020-03-21:
Now its Works on my rgb 15khz.

closed 2020-03-21 by Julitium