mist-devel/mist-board #143: Firmware: exFAT support

solskogen, 2020-04-14:
Sometimes it would be awesome to have files bigger than 4GB on the SD card (big hdf's for example), but that is not possible at the moment.

jotego, 2020-04-17:
I think it will be hard to find an exFAT implementation for the microcontroller.

harbaum, 2020-04-17:
Also the SD card in the mist is rather slow. So there's basically no valid use case for gigabytes of data. It would take the mist several days to read that. And if you only use a fraction of the image you can likely live with less than 2gb max image size.

solskogen, 2020-04-17:
It works quite well on MiSTer when I have hdf files that are 8GBs (for the
whole WHDL collection)

gyurco, 2020-04-17:
With direct SD access, it's about 1 megabytes/sec. With sequential read you could read 8 GiB in 8000 secs. Random reads and writes will take forever.
Btw, I don't think there's enough Flash/RAM for an exFAT implementation.

gyurco, 2021-05-15:
If still interested:
https://www.atari-forum.com/viewtopic.php?f=115&t=40499

gyurco, 2021-05-30:
Latest firmware supports the exFAT filesystem.

closed 2021-05-30 by gyurco