mist-devel/mist-board #148: [BBC] Video problems with PVM and new core

squidrpi, 2020-05-26:
With the latest core the video is jittery on a PVM monitor. Previous version, i.e. 180823 was OK. Newer version the image is brighter.

This new version fixes the sound timing and tuning issues in Bad Apple and Beebtracker!

gyurco, 2020-05-26:
You can try to revert to the old CRTC to see if it's the problem. On my display, the picture is stable.

squidrpi, 2020-05-26:
Picture is stable on consumer TV RGB CRT but the menu usually is jittery.

nippur72, 2020-05-26:
on my CRT TV (composite output), the picture is stable but I can't see any color, all is B/W. But I don't see differences from the old core, it's the same.

Anyway I think I know what might be the issue regarding colors at least -- it once happened with VIC20 and C16. It was the csync signal calculated as `HSYNC ^ VSYNC` instead of `~(~HSYNC | ~VSYNC)`. Perhaps the two issues are related.

gyurco, 2020-05-27:
In all modes?
Try with the BASIC command MODE 0-1-2-3-4-5-6-7

gyurco, 2020-05-27:
The csync formula is the same as in the old core, but I might add the second version (basically it's an AND relation) to mist-video, selectable via a parameter.

gyurco, 2020-05-27:
Did some tweaking of the csync signal, can you test it again from the repo? The OSD is not interlace-friendly, that will still look a bit strange. Might fix it later (not a high prio for me).

nippur72, 2020-05-27:
I tested the new csync method there is an improvement, the image is brighter, I can see some colors but I think they are not correct. I can only see white, black, yellow and red. Blue color sometimes is flickering heavily and some other times is gray.

Now I can see the effect of interlacing -- the image it's flickering to a certain extent, especially when the OSD menu is on.

squidrpi, 2020-05-27:
gyurco, can you update the Quartus files? I'm getting compile errors with the current git files.

nippur72, 2020-05-27:
@squidrpi I compiled it without errors, I just cloned a fresh repo and then opened it into Quartus.

squidrpi, 2020-05-27:
No difference on the PVM for me.

squidrpi, 2020-05-27:
May just be a problem with PVMs and 50hz interlaced - they're not great with non 240p signals. I've noticed shakey images on other cores that do interlaced. Picture looks fine on a consumer RGB CRT so I would ignore this if even though the older core displayed fine on the PVM.

closed 2020-05-27 by squidrpi

gyurco, 2020-05-27:
The older core didn't do interlace. It might be good or bad, depends on the point of view.