mist-devel/mist-board #149: [C64] EasyFlash .crt 1MB compilations not working now.

palehub, 2020-05-29:
Hi. I have recently updated to the latest core for the C64 (200522).
I was using the prior release (200430).
The ROM I use normally is the one that contains the Jiffy DOS, but I have also tried with the normal ROM.

I have noticed that before the last version I could load all the CRT cartridges of the EasyFlash normally and they worked fine, but with the new last core update the cartridges (.crt) that contain large compilations (those of 1MB) do not load. The screen remains black. Smaller compilations still works fine. It seems a matter of size, I think.

I expose it here to put on record in case someone notices the same.

gyurco, 2020-05-29:
Is there any .crt sample file which I can try?

palehub, 2020-05-29:
> Is there any .crt sample file which I can try?

Yes. Tank You. I uploaded an example. 1MB .crt compilation with the mentioned problem.

https://drive.google.com/file/d/1xkJ0GtaCVbFqd8LznLs5v7KywJKlgxlS/view?usp=sharing

gyurco, 2020-05-29:
It works for me. Maybe it's something similar to https://github.com/mist-devel/mist-board/issues/147.
Can you try it with other SD Card?

gyurco, 2020-05-29:
BTW, the latest release was 200430. If you're using a build from somewhere else, then ask who built that RBF.

closed 2020-05-29 by gyurco

palehub, 2020-05-29:
> It works for me. Maybe it's something similar to #147.
> Can you try it with other SD Card?

It works for me also on release 200430, if that is also the release you have tried it is no surprise.

palehub, 2020-05-29:
> BTW, the latest release was 200430. If you're using a build from somewhere else, then ask who built that RBF.

Yes. I thought release 200522 was official. The place where I downloaded It pointed me to post my issue here. Sorry if that is not correct.

gyurco, 2020-05-29:
If it's not downloaded from mist-binaries, then please go to the guy who gave it to you. I think it's fair.

palehub, 2020-05-29:
> If it's not downloaded from mist-binaries, then please go to the guy who gave it to you. I think it's fair.

I agree.

palehub, 2020-05-29:
> If it's not downloaded from mist-binaries, then please go to the guy who gave it to you. I think it's fair.
> Nothing was fixed after 200430, so your version either has some custom modification, or the person who built is simply clueless.

Probably. I have been misinformed. Sorry.