mist-devel/mist-board #150: Add ability to avoid db9 joystick renumbering (per core option)

PunisherHD92, 2020-06-05:
Hello, I currently use my mist with following devices connected:
1 * USB mouse
2 * DB9 joystick
1 * USB joypad

Currently, the USB pad is always seen as joystick 1, whereas DB9 joysticks are seen as joysticks 2 and 3.

I'd like to be able to force some specific core (mostly computer like amiga, atarist, c64 ...) to always use DB9 joystick first, and keep other core (console like megadrive ...) to always use the usb pad first.

I've tried the 'joystick0_prefer_db9' option but it does not do what I want.
The option could be named something like 'joystick_db9_fixed_index' and located in core-specific sections of mist.ini

gyurco, 2020-06-05:
Almost all cores have a swap joystick option (well, Amiga and ST are the exceptions).

PunisherHD92, 2020-06-05:
Hello, to my knowledge, the swap option allows to swap the 2 first logical joystick ports, not physical ports.

E.g. when running the C64 core, by default it sets:
joystick1=usb, joystick2=db9(1), joystick3=db9(2) .

If I enable joystick swap, it  swaps joystick1 and 2, so it sets!
joystick1=db9(1), joystick2=usb, joystick3=db9(2)

This is not what I need. What I need is:
joystick1=db9(1), joystick2=db9(2), joystick3=usb

gyurco, 2020-06-05:
That's true. But at least you can use the USB gamepad, and also both DB9 joystick, if you disconnect the USB. With the forced DB9 preference, it won't be possible to use USB without changing mist.ini. Maybe it's not a problem.

PunisherHD92, 2020-06-05:
You are right, a very simple solution is to unplug the USB pad when I don't want to use it :-)

What I meant in my original post was a mist.ini like this:
```
[mist]
joystick_db9_fixed_index=0 ; Use USB joystick if available, else fallback to DB9

[c64]
joystick_db9_fixed_index=1 ; Use DB9 joysticks even if USB joystick is available
```
I'm writing a patch to implement this option in firmware; currently it works when Mist boots straight to the c64 core; but it does not when Mist boots to menu-core then switch to c64 core. It looks like the USB devices are scanned and joystick numbering calculated when Mist is powered-on, not when a new core is started.

gyurco, 2020-06-05:
More likely it's done when the joypad is plugged/unplugged. To achieve your goal, the renumbering logic should be completely removed from hid.c, and handle it only in menu and user_io.

PunisherHD92, 2020-06-06:
Hello, i've managed to make it work as I expected. You can see the pull request here:
https://github.com/mist-devel/mist-firmware/pull/55

closed 2020-06-09 by gyurco