mist-devel/mist-board #152: Ready-to-use USB-Rtc

uchristo, 2020-07-27:
Will there ever be a complete rtc module for Mist. I saw that firmware now adds a general support. I‘d lobe to have some kind of dongle with just a 3V-Battery to connect.

gyurco, 2020-07-27:
It's better to ask manufacturers, like Lotharek or Manuferhi. DIY method is not that hard actually, maybe programming the ATtiny is a bit uncomfortable.
Btw, I just ordered 10 PCBs, I'll build one for myself, probably I'll build the 9 others to sell.

uchristo, 2020-07-27:
My goal would be to have both (USB->i2c & rtc) on one PCB. Battery-clip could be mounted vertically so one could use one of the upper USB Ports Without blocking another port.

Coule the rrc be connected to rhe seri

gyurco, 2020-07-27:
In theory you can connect the DS1307 to the UART port directly, and drive the SCL and SDA from the FPGA. But no core implements that, since the USB connection is universal.

gyurco, 2020-07-27:
My PCB will have the MCU, the RTC and the battery clip attached, with an USB-B connector. I didn't want to make a "pendrive" like dongle, since with the battery clip, the PCB is a bit long.

uchristo, 2020-07-27:
> My PCB will have the MCU, the RTC and the battery clip attached, with an USB-B connector. I didn't want to make a "pendrive" like dongle, since with the battery clip, the PCB is a bit long.

Can you contact me for one of the 9 PCBs? > uwe {at} christof {dot} cc

gyurco, 2020-07-27:
Of course, when it'll be ready.

harbaum, 2020-07-28:
You can also build these from digisparks: http://digistump.com/products/1 

Maybe you want to do a similar design incl the RTC.

gyurco, 2020-07-28:
And with such miniature design I would probably make the RTC part sit on top of the USB-I2C, this way it would not stick out more than 3 cm in a dongle approach. However my "long" PCBs already shipped, I will post some pictures on the forum if they arrive.

gyurco, 2020-07-28:
Another look on the MiST: as the USB ports are very close to each other, the dongle approach with a horizontal battery holder will have a problem that it won't allow plugging anything into the next port. A vertical battery holder - I wouldn't like it, because it's too tall. Using smaller battery, not CR2032 - maybe.
I think I'll be happy with the separate box next to the MiST, connected with an USB A-B cable.

harbaum, 2020-07-28:
A cr1632 at the very end of the pcb may be nice.

robinsonb5, 2020-07-28:
For what it's worth, the Chameleon64's RTC battery is a CR1220.  That should fit reasonably well in pendrive format.

gyurco, 2020-07-28:
Good ideas for a dongle-type V2. Just for me, it will be tricky to program an SMD ATTiny45. Now using an Arduino as ISP and a breadboard :)

harbaum, 2020-07-28:
One of the super cheap ways i use for this is a USBASP like:
https://de.aliexpress.com/item/32670511994.html?spm=a2g0o.productlist.0.0.1c0c200d1qQUXx
in conjunction with this:
https://de.aliexpress.com/item/32838153476.html?spm=a2g0s.9042311.0.0.9a654c4da9dV2o

As a result I can program them in place without adding any programming pins to the board.

gyurco, 2020-07-30:
Interesting. I can not really imagine how this clip works, but it's really cheap, so I'll try it. I guess that programmer cannot do HVSP programming (but it's not really important anyway).

gyurco, 2020-08-07:
@uchristo I uploaded some pictures to the forum:
https://www.atari-forum.com/viewtopic.php?f=115&t=39302
If you still want one, PM me.

closed 2021-07-18 by gyurco