mist-devel/mist-board #156: >4GB support for Amiga SD partitions

gcopoix, 2021-04-11:
MiST Minimig AGA/RTG offers support for direct partition access.
If these partitions are marked with partitionID 0x30 or 0x76 they show up in WinUAE too.

Only problem: >4GB (?) MBR partitions are not supported (seems to me).

I've tried an 8GB partition on an 32GB SDHC card (partition ID=0x76 or 0x30), this partition shows up in WinUAE (must be started with admin rights).
If I now create 2 Amiga Partitions (2GB FFS and 6GB PFS) they work seamless in WinUAE, but the PFS partition shows up in MiST as NDOS. Formatting or setting up the 6GB partition within MiST fails too.

gyurco, 2021-04-11:
I have a (secondary) 14G partition which works well with Minimig AGA-RTG, so the 4G limit is surely not there.
I'm using FAT95 with it, but it's just the software side, on the hardware side, there's no such limit.
I don't know if it works with WinUAE, with FAT95, it doesn't need to be available in an emulator to be able to copy files on it.

gcopoix, 2021-04-11:
Hi, with fat95 i am able to access bigger partitions too, my question was regarding SD Partitions presented to minimig_aga as "native" amiga HDDs (so I am able to partition from Amiga side as the fat95 doesn't fully support all Amiga filesystem features)

gyurco, 2021-04-11:
Maybe I wasn't clear, but the FAT partition is presented to the core as MMC/SD partition 2. If you experience a limit, it's not in the hardware.

gcopoix, 2021-04-11:
Maybe to make it more understandable:
- 32GB SD card has 2 partitions:
   1. 20GB FAT32 as primary partition
   2. 8GB unformatted primary partition with a partition ID = 0x76
- the 8GB partition, mounted in WinUAE as "Add Harddrive" and choosed "[PART] 7.9GB P#1 ...." in "Full drive/RDB mode"
- in Amiga configure this "HD" by HDInstTools with 2 Amiga partitions, 2G (ffs) and 6G (pfs)
   works perfectly in WinUAE
- back to MiST only 2G Amiga Partition is valid, the 6G Amiga partition shows up as NDOS
- the same with an 4G partition (divided into 2x 2GB Amiha partitions) work perfect on MiST too,
  only if this SD-Partition is >4GB the higher partitions doesn't work anymore.

So I assume there must be some bug in scsi.device (tried differnet 43.xx and 44.xx) <-> Core <-> MiST firmware handling

gyurco, 2021-04-11:
Strange, because the IDE driver doesn't synthesize a Fake RDB for partitions, it just serves block 0 as-is on the SD Card (partition). Maybe @robinsonb5 has an idea.
Examining the RDB doesn't show any issues?

gcopoix, 2021-04-11:
RDB of this partition shows 16381/16/63, so should fit into all variables and is below the 7,87GB barrier. So for me there is no issue visible.
The question to fully support the TD64 commands comes next ;-)

robinsonb5, 2021-04-11:
Do you definitely have the filesystem installed within the RDB?  (In some circumstances WinUAE will add the filesystem either via a ROMTag or by injecting into the RDB on the fly.  I don't *think* this is one of those scenarios, but it's worth checking.)

I think I have a program somewhere for RDB / partition analysis which I wrote years ago for use with Zip drives - I'll see if I can find it.

gyurco, 2021-04-11:
FAT95 can deal with it somehow, but I'm not an expert on Amiga drivers/filesystems.

gyurco, 2021-04-11:
As I see, LBA bit is not respected in the firmware, only CHS mode is used. Might be an issue?

gcopoix, 2021-04-11:
> Do you definitely have the filesystem installed within the RDB?

Yes - both FFS (updated) and PFS

robinsonb5, 2021-04-15:
I'll see if I can reproduce this - so you happen to know whether this is PFS5.3 or PFSAIO?

gcopoix, 2021-04-16:
It is the AiO version (from [ClassicWB](http://classicwb.abime.net/classicweb/full.htm) (MyFiles/LargeHD/... or something)

gcopoix, 2021-06-08:
seems fixed since in firmware_210530.

closed 2021-06-08 by gcopoix